tphakala/birdnet-go #36: Multiple nodes, single MySQL database

PeteLawrence, 2024-02-21:
I have setup a single MySQL database with 2 BirdNet-Go nodes sending their detections to that database. I noticed that there's a source_node field in the database so thought the web interface for each node would show only their respective detections, however the web interface on both nodes shows all detections, along with broken spectrograms for detections from the other node as the wav/png file doesn't exist on that node.

Should the web interface show only that nodes detections, or is it meant to show detections from all nodes?

tphakala, 2024-02-21:
I have planned to implement support multinode setup, node name and shared database support is part of that, but it also is all I have done about it yet. So any plans and implementation further is still to be done.

PeteLawrence, 2024-03-17:
Would you be interested in a pull request that adds in station name filtering to the relevant DB queries?