tphakala/birdnet-go #37: Permission error when running as systemd service

farski, 2024-02-27:
This could be something very obvious that I'm missing, I don't deal with systemd all that often.

My system has a user called `birdnet`. I have created the following systemd service

```
[Unit]
Description=BirdNET-Go realtime analysis
After=network.target
StartLimitIntervalSec=0

[Service]
Type=simple
Restart=always
RestartSec=5
User=birdnet
ExecStart=/home/birdnet/birdnet-go/birdnet-go realtime

[Install]
WantedBy=multi-user.target
```

When I try to start that service, I get 

```
Feb 27 13:30:43 birdnet systemd[1]: Started birdnet-go.service - BirdNET-Go realtime analysis.
Feb 27 13:30:43 birdnet birdnet-go[110182]: Read config file: /home/birdnet/.config/birdnet-go/config.yaml
Feb 27 13:30:43 birdnet birdnet-go[110182]: BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
Feb 27 13:30:43 birdnet birdnet-go[110182]: Starting BirdNET-Go Analyzer in realtime mode
Feb 27 13:30:43 birdnet birdnet-go[110182]: Threshold: 0.8, sensitivity: 1, interval: 15
Feb 27 13:30:43 birdnet birdnet-go[110182]: 2024/02/27 13:30:43 rename webui.log webui.log.20240227133043: permission denied
Feb 27 13:30:43 birdnet systemd[1]: birdnet-go.service: Main process exited, code=exited, status=1/FAILURE
Feb 27 13:30:43 birdnet systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
```

I have tried also moving the installation to `/opt/birdnet-go` and checked permissions on all relevant files and directories. It seems like no matter what I do, running the program as the `birdnet` user directly always works, and running it via systemd as the `birdnet` user always fails with this renaming error. Also notable is that it seems to attempt to rename the `webui.log` file, even if there is no existing `webui.log` file in the same directory as `birdnet-go`.

I am able to work around this by providing an absolute path for the webserver log path in `config.yaml`. I think I would generally assume that the paths specified in the config are always relative to the program binary, but maybe that's a bad assumption.

Once I get past the renaming error, I get a new error:

```
Feb 27 13:43:36 birdnet systemd[1]: Started birdnet-go.service - BirdNET-Go realtime analysis.
Feb 27 13:43:36 birdnet birdnet-go[110371]: Read config file: /home/birdnet/.config/birdnet-go/config.yaml
Feb 27 13:43:36 birdnet birdnet-go[110371]: BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
Feb 27 13:43:36 birdnet birdnet-go[110371]: Starting BirdNET-Go Analyzer in realtime mode
Feb 27 13:43:36 birdnet birdnet-go[110371]: Threshold: 0.8, sensitivity: 1, interval: 15
Feb 27 13:43:37 birdnet birdnet-go[110371]: [2024-02-27T13:43:37-05:00] [INFO] ⇨ http server started on [::]:8080
Feb 27 13:43:37 birdnet birdnet-go[110371]: ALSA lib pcm_dsnoop.c:566:(snd_pcm_dsnoop_open) unable to open slave
Feb 27 13:43:37 birdnet birdnet-go[110371]: ALSA lib pcm_dsnoop.c:566:(snd_pcm_dsnoop_open) unable to open slave
Feb 27 13:43:37 birdnet birdnet-go[110371]: ALSA lib pcm_dsnoop.c:566:(snd_pcm_dsnoop_open) unable to open slave
Feb 27 13:43:37 birdnet birdnet-go[110371]: 2024/02/27 13:43:37 Device init failed miniaudio: Failed to open backend device
Feb 27 13:43:37 birdnet systemd[1]: birdnet-go.service: Main process exited, code=exited, status=1/FAILURE
Feb 27 13:43:37 birdnet systemd[1]: birdnet-go.service: Failed with result 'exit-code'.
```

I don't have a good idea for how to deal with that one.

But the tl;dr is that something seems to be handled differently when running as a user versus running as a user via systemd, which leads to a number of problems.

tphakala, 2024-02-27:
All paths in birdnet-go config are relative by default, you need to set working directory in systemd config unless you configure absolute paths in config (like path: /opt/birdnet-go/logs/webui.log).

This is my /etc/systemd/system/birdnet-go.service

```
[Unit]
Description=BirdNET-Go Analyzer Service

[Service]
ExecStart=/home/thakala/birdnet-go realtime -d
Restart=on-failure
User=thakala
Group=thakala
WorkingDirectory=/home/thakala

[Install]
WantedBy=multi-user.target
```

To fix ALSA permission issue you need to add birdnet user as member of audio group

`sudo usermod -a -G audio birdnet`

farski, 2024-03-01:
I have rebooted, and `alsamixer` does work for the birdnet user (or at least it launches and I can change the volume

[image]

farski, 2024-03-04:
@tphakala Thanks for all that testing! The computer where I'm currently running BirdNET-Go actually does **not** have BirdNET-Pi on it anywhere. This was a fresh OS with only BirdNET-Go installed.

I do see `pulseaudio` running in top, though. In the other thread I know you said you disable Pulse Audio for BirdNET-Go, so perhaps it's a similar issue? Though when I have tried running BirdNET-Go through systemd, I always stop any other running instances first, so I wouldn't expect them to be competing. But I don't think I've ever tried running it via systemd after a restart without having run it manually at all, so maybe something is hanging around. I can test that.

But also maybe once I'm on a version that doesn't use Pulse Audio at all it won't matter?

tphakala, 2024-03-04:
I disabled Pulse Audio because it is causing inconsistent behavior due to how Pulse Audio works by default. When you login to Linux shell, by SSH or by any other means, this will typically also spawn a Pulse Audio server, once you log out Pulse Audio server stops.

Miniaudio library I am using in BirdNET-Go connects to Pulse Audio server if it is available, if it is not it falls back to ALSA. In your case Pulse Audio backed audio works, but ALSA does not (hence errors with systemd launched process).

Could you try to add following config into /home/birdnet/.asoundrc (or to profile you are running BirdNET-Go as)

`pcm.capture {
    type dsnoop
    ipc_key 5678
    slave {
        pcm "hw:0,0"
        channels 2
        rate 48000
        format S16_LE
    }
}

pcm.!default {
    type hw
    capture.pcm {
        type plug
        slave.pcm "capture"
    }
}`

This will configure ALSA to provide audio capture device as dsnoop type device which can be shared between processes.

tphakala, 2024-03-04:
Is your system a full desktop install or is it a minimal install?

farski, 2024-03-04:
> Is your system a full desktop install or is it a minimal install?

I believe it was a vanilla Raspberry Pi OS Lite 64 bit install from a few weeks ago

jetpacktuxedo, 2024-05-07:
@farski idk if you are still having this issue, but I was seeing the same thing and fixed it by adding
```
ExecStartPre=/usr/bin/pulseaudio --start
```
to my systemd script. I _think_ the pulseaudio server doesn't get started for systemd sessions, and this triggers it to start up before it kicks off the normal startup process.

There might be a better way to do this because linux audio is far from my strong suit, but it seems to be working for me 😅