tphakala/birdnet-go #45: Documenting Web UI Usage

arnydo, 2024-03-05:
Hi! Fantastic project BTW!

I am setting up an instance here and I am unable to determine how to get the Web UI running. Do you have any documentation on setting this up?

Thank you.

arnydo, 2024-03-05:
Hey @tphakala  thank you so much for the prompt response.

I am running the same as you but it isn't loading the UI.
It is the default config running on Linux.

```
./birdnet-go realtime          
Read config file: /home/arnydo/.config/birdnet-go/config.yaml
error reading species conficende config: species confidence file 'species_config.csv' not found
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 24 threads of available 24 CPUs
Starting BirdNET-Go Analyzer in realtime mode
Threshold: 0.8, sensitivity: 1, interval: 15
Listening ...
11:02:21 Engine 0.86
```

tphakala, 2024-03-05:
According to that output web server is not starting.

Can you check that this section is included in /home/arnydo/.config/birdnet-go/config.yaml?

```
webserver:
  enabled: true		# true to enable web server
  port: 8080		# port for web server
  autotls: false		# true to enable auto TLS
  log:
    enabled: true	# true to enable log file
    path: webui.log	# path to log file
    rotation: daily	# daily, weekly or size
    maxsize: 1048576	# max size in bytes for size rotation
    rotationday: 0	# day of the week for weekly rotation, 0 = Sunday
```

If you have used version of BirdNET-Go without web ui this is likely missing from config, I have not yet implemented any method for updating config across versions 😞 

If you delete existing config file a new config with new default values will be created for you, just make sure to make a note of important settings you may have done.

arnydo, 2024-03-05:
Hmm. Still not working. I downloaded [v0.4.4](https://github.com/tphakala/birdnet-go/releases/tag/v0.4.4) instead of compiling the latest. Could that be the issue?

tphakala, 2024-03-05:
Yes! That is indeed the issue, this is only binary release with web ui included at the moment https://github.com/tphakala/birdnet-go/releases/tag/20240215

But there is also Docker images now, and these are currently build based on latest code in Github https://github.com/tphakala/birdnet-go/pkgs/container/birdnet-go, however this container model is very new to this app and things are not yet very user friendly with configuration etc. with that. But Docker container will be preferred deployment model going forward.

closed 2026-02-14 by tphakala