tphakala/birdnet-go #48: Docker image fails to upload soundscapes to BirdWeather

PeteLawrence, 2024-03-10:
When posting soundscapes to BirdWeather when running the birdnet-go Docker image I get the following error -

`2024/03/09 17:42:45 Request to upload soundscape failed: Post "https://app.birdweather.com/api/v1/stations/XXXXXXXXXX/soundscapes?timestamp=2024-03-09T17:42:42.000+0000": tls: failed to verify certificate: x509: certificate signed by unknown authority`

I believe this can be resolved by adding the `ca-certificates` package to the image.

tphakala, 2024-03-10:
Thanks for reporting, fixed in #49

tphakala, 2024-03-11:
There is still a known issue that WAV files uploaded to Birdweather are encoded incorrectly so they contain just silence

tphakala, 2024-10-16:
This has been fixed

closed 2024-10-16 by tphakala