tphakala/birdnet-go #67: Connecting to non-existing rtsp stream spawns lots of ffmpeg processes

isZumpo, 2024-03-18:
Got around to trying out the rtsp streaming feature and it works really well... Once you have filled in the correct url!

The scenario can be illustrated by starting the server while specifying a rtsp server that does not exist:

> realtime --rtsp rtsp://madeupserverthatdoesnotexist.com:8554/mystream --rtsptransport tcp

Once started, the server will try to reconnect over and over again. Each time spawning a new ffmpeg process.

**Console output snippet**
```bash
2024/03/18 20:25:15 Starting ffmpeg with command:  /usr/bin/ffmpeg -rtsp_transport tcp -i rtsp://madeupserverthatdoesnotexist.com:8554/mystream -loglevel error -vn -f s16le -ar 48000 -ac 1 pipe:1
Restarting audio capture
2024/03/18 20:25:16 Error reading from ffmpeg: EOF
2024/03/18 20:25:16 Starting ffmpeg with command:  /usr/bin/ffmpeg -rtsp_transport tcp -i rtsp://madeupserverthatdoesnotexist.com:8554/mystream -loglevel error -vn -f s16le -ar 48000 -ac 1 pipe:1
2024/03/18 20:25:16 Error reading from ffmpeg: EOF
Restarting audio capture
2024/03/18 20:25:16 Starting ffmpeg with command:  /usr/bin/ffmpeg -rtsp_transport tcp -i rtsp://madeupserverthatdoesnotexist.com:8554/mystream -loglevel error -vn -f s16le -ar 48000 -ac 1 pipe:1
2024/03/18 20:25:16 Error reading from ffmpeg: EOF
Restarting audio capture
2024/03/18 20:25:16 Starting ffmpeg with command:  /usr/bin/ffmpeg -rtsp_transport tcp -i rtsp://madeupserverthatdoesnotexist.com:8554/mystream -loglevel error -vn -f s16le -ar 48000 -ac 1 pipe:1
2024/03/18 20:25:17 Error reading from ffmpeg: EOF
Restarting audio capture
2024/03/18 20:25:17 Starting ffmpeg with command:  /usr/bin/ffmpeg -rtsp_transport tcp -i rtsp://madeupserverthatdoesnotexist.com:8554/mystream -loglevel error -vn -f s16le -ar 48000 -ac 1 pipe:1
2024/03/18 20:25:17 Error reading from ffmpeg: EOF
...
... Just keeps going...
...

```
**Htop view**
[image]

The processes will very quickly pile up and bring your computer down (my poor old laptop server :( ). I'm now setting a cpu limit for BirdNet-Go in my deployments to be on the safe side.

I have tried with your most recent [change](https://github.com/tphakala/birdnet-go/commit/4652ff349f280ac3beb7ebf65225e4d065d55be9) and the behavior is the same there too.

tphakala, 2024-03-18:
I missed some error cases then, I will fix this later

tphakala, 2024-03-19:
fixed with 4b7faf0

isZumpo, 2024-03-19:
Just verified the most recent commit. It now works as expected. Thanks for a quick fix and great work!

closed 2024-03-19 by isZumpo