tphakala/birdnet-go #69: Implement audio file purging

tphakala, 2024-03-21:
Old wav files should be purged once disk space is starting to get low, purging should be flexible so that users can select which species recordings should be deleted last.

isZumpo, 2024-04-06:
Should only the audio file be removed or the whole database entry for a given detection?

MueJosh, 2024-04-07:
> Should only the audio file be removed or the whole database entry for a given detection?

Definitely just the audio files. (It makes much more sense this way)

isZumpo, 2024-04-07:
> > Should only the audio file be removed or the whole database entry for a given detection?
> 
> Definitely just the audio files. (It makes much more sense this way)

Agreed! The database entries should be of so little size that it does not make any sense to clear them up in comparison to the clips. They could also be useful for long term statistics and insights.

tphakala, 2024-10-16:
Audio file purging has been implemented