tphakala/birdnet-go #83: Feature request: mqtt publishing of detections

tomlut, 2024-03-26:
Could you please consider publishing bird detections to an mqtt broker?

tphakala, 2024-04-05:
MQTT support is now added, code was contributed by @janvrska. Release v0.5.1 and latest docker images contain support for this, please test and provide feedback.

Add following under realtime node in config yaml file

    realtime:
        mqtt:
            enabled: true                      # true to enable MQTT
            broker: tcp://localhost:1883       # MQTT (tcp://host:port)
            topic: birdnet                     # MQTT topic
            username: birdnet		   # MQTT username
            password: secret       		   # MQTT password

tomlut, 2024-04-05:
Confirmed as working. Thank you both.

birdnet startup
```
Starting BirdNET-Go Analyzer in realtime mode
Threshold: 0.8, sensitivity: 1, interval: 15
2024/04/06 08:22:42 Connecting to MQTT broker
2024/04/06 08:22:42 Successfully connected to MQTT broker
2024/04/06 08:22:42 Connected to MQTT broker: tcp://10.1.1.100:1883
[2024-04-06T08:22:43+11:00] [INFO] â¨ http server started on [::]:8080
```

mqtt broker /birdnet topic:
```
{
  "ID": 0,
  "SourceNode": "BirdNET-Go",
  "Date": "2024-04-06",
  "Time": "08:23:51",
  "InputFile": "",
  "BeginTime": "2024-04-06T08:23:48.736132138+11:00",
  "EndTime": "0001-01-01T00:00:00Z",
  "SpeciesCode": "railor5",
  "ScientificName": "Trichoglossus moluccanus",
  "CommonName": "Rainbow Lorikeet",
  "Confidence": 0.8614452481269836,
  "Latitude": <redacted>
  "Longitude": <redacted>
  "Threshold": 0.8,
  "Sensitivity": 1,
  "ClipName": "clips/2024/04/trichoglossus_moluccanus_86p_20240406T082353Z.wav",
  "Comment": "",
  "ProcessingTime": 598766979,
  "Results": null
}
```

Just have to work out how best to display it in Home Assistant now.

closed 2024-04-05 by tomlut