tphakala/birdnet-go #90: panic: runtime error: invalid memory address or nil pointer dereference

Pangslinger, 2024-03-29:
When running birdnet-go I get following error. I've documented the steps of my installation here, in case I missed something: https://github.com/tphakala/birdnet-go/discussions/82#discussioncomment-8949636

birdnet-go realtime
Read config file: /home/username/.config/birdnet-go/config.yaml
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
BirdNET locale not set, using English as default
Starting BirdNET-Go Analyzer in realtime mode
Threshold: 0.8, sensitivity: 1, interval: 15
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x40 pc=0x9b5698]

goroutine 1 [running]:
github.com/tphakala/birdnet-go/internal/analysis.RealtimeAnalysis(0x40001ec240)
        /home/thakala/go/src/github.com/tphakala/birdnet-go/internal/analysis/realtime.go:47 +0x1f8
github.com/tphakala/birdnet-go/cmd/realtime.Command.func1(0x40001e4e00?, {0xcca26a?, 0x4?, 0xcca1a2?})
        /home/thakala/go/src/github.com/tphakala/birdnet-go/cmd/realtime/realtime.go:20 +0x20
github.com/spf13/cobra.(*Command).execute(0x400011cf00, {0x54f6ba0, 0x0, 0x0})
        /home/thakala/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:983 +0x840
github.com/spf13/cobra.(*Command).ExecuteC(0x400011cc00)
        /home/thakala/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1115 +0x344
github.com/spf13/cobra.(*Command).Execute(0x40000466e8?)
        /home/thakala/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1039 +0x1c
main.main()
        /home/thakala/go/src/github.com/tphakala/birdnet-go/main.go:33 +0xd8

tphakala, 2024-03-29:
How did you install birdnet-go, did you compile it from sources or did you download one of releases?

tphakala, 2024-03-30:
Can you test latest release v0.5.0?

Pangslinger, 2024-03-30:
Not at the moment, sorry. I've already installed the classic birdnet version and want to keep it for now. But I will buy another pi and check it out as soon as possible!

closed 2024-06-11 by tphakala