tphakala/birdnet-go #96: Location Filter Threshold is fixed but should be set by User

MueJosh, 2024-04-01:
Hey, 

I dug up some stuff from BirdNET 
It is really important that the user of BirdNET-Go can set the locationFilterThreshold for themselves! 

See line 62
[BirdNET-Analyzer; config.py](https://github.com/kahst/BirdNET-Analyzer/blob/main/config.py)

I think you defined it in Line 33 
[rangefilter.go](https://github.com/tphakala/birdnet-go/blob/main/internal/birdnet/rangefilter.go)

If the threshold is fixed (like it is right now) I can't detect certain species that are definitely present on my location. 

On BirdNET-Pi (Nachtzuster Fork- but its basically the same) it looks like this: 
[image]
The link from the photo
[here](https://github.com/mcguirepr89/BirdNET-Pi/wiki/BirdNET-Pi:-some-theory-on-classification-&-some-practical-hints)

There is also a small species list tester- so i know where i set my threshold 
For the extreme i set it to 0.95 - see Picture below
[image]
And for my location I have it set to 0.0018 so i can detect most owls in my region. 
As you can see the list gets MUCH longer 
[image]

### This is not needed for the person that just wants to know their birds in the garden. But for people that are interested in ornithology and Enthusiasts (like me) it is Really important! 
## Otherwise a lot of birds don't get detected even tho they are present, calling and are in the birdnet library.

tphakala, 2024-04-02:
Implemented with commit 954c681

Example of configuring a value in config.yaml

    birdnet:
      locationfilterthreshold: 0.001

If you use docker images pull latest image by running `docker pull ghcr.io/tphakala/birdnet-go:latest`

MueJosh, 2024-04-02:
> Implemented with commit 954c681

Great! 

But there is still something wrong with the threshold / location 

#### I have following settings: 
Config.yaml
```
...
# BirdNET model specific settings
birdnet:
    sensitivity: 1.25 # sigmoid sensitivity, 0.1 to 1.5
    threshold: 0.7 # threshold for prediction confidence to report, 0.0 to 1.0
    overlap: 0.5 # overlap between chunks, 0.0 to 2.9
    threads: 4 # 0 to use all available CPU threads
    locale: de # language to use for labels
    latitude: 48.XXXX # latitude of recording location for prediction filtering
    longitude: 7.XXXX # longitude of recording location for prediction filtering
    locationfilterthreshold: 0.0018 # rangefilter species occurrence threshold
...
```
Species_config.csv 
```

# Example configuration for species-specific confidence thresholds
# Format: species, confidence
# - Species: Localized common name or scientific name (case insensitive)
# - Confidence value: Range from 0.0 to 1.0 for species reporting.
#   Values exceeding 1.0 are used to explicitly exclude species

# Exclude non-bird labels (e.g., environmental sounds, human activities)
Dog,2.0
Engine,2.0
Environmental,2.0
Fireworks,2.0
Gun,2.0
Human non-vocal,2.0
Human vocal,2.0
Human whistle,2.0
Noise,2.0
Power tools,2.0
Siren,2.0

# Higher threshold for uncommon species (which are frequently misidentified)
common kingfisher,0.85
eurasian moorhen,1.0
common snipe,1.0

# Lower threshold for owls (due to rarity or specific interest)
western barn owl,0.25
boreal owl,0.25
eurasian eagle-owl,0.25
eurasian pygmy owl,0.25
eurasian scops owl,0.25
little owl,0.25
long-eared owl,0.25
northern hawk-owl,0.25
short-eared owl,0.25
snowy owl,0.25
tawny owl,0.25

```

And I get some weird detections that really shouldn't be detected. With the species list tester set to my location and a threshold of 0.0018 there shouldn't be a "flammulated owl" (native to America; I'm in Germany^^) 

Maybe it's because the variable can have 4 decimals instead of the 2 that are everywhere else. 
Or it now overwrites the location somehow and gets the LocationFilterThreshold for the whole world. (Which is basically 0.000(5) no matter the defined LFT)

[image]
Ignore the noisy PSU... it works for testing purposes just fine

closed 2024-04-03 by MueJosh