tphakala/birdnet-go #99: #bug species_config.csv just works with language set to English

MueJosh, 2024-04-04:
I set up the species config and it only works if I have birdnetgo set to English. (In config.yaml)

I tried the common names for German (inside the species config) but that doesn't change the output (NO Output) 

This works:
	Config.yaml 
	Language = en 

	Species_config.csv
	BirdNames in English [ tawny owl,0.2

This does not work:
	Config.yaml 
	Language = de 

	Species_config.csv
	BirdNames in English 
	(neither with German names [ waldkauz,0.2 ])

tphakala, 2024-04-05:
I have species_config.cvs configured with Finnish labels and they do work, there is something else going on now. Can you attach your full species_config.cvs to this issue so can I debug with it?

MueJosh, 2024-04-05:
Mhmm, that’s weird
So you're using common name? 

Sure 
[species_config.csv # added info](https://github.com/tphakala/birdnet-go/files/14885520/species_config.csv)

closed 2026-02-14 by tphakala