tphakala/birdnet-go #105: Dasboard displays yesterday

tomlut, 2024-04-05:
When opening the web dashboard I get yesterday's data displayed and have to use the date picker to select today.  Also when reloading the browser window I get yesterday (Firefox, on Windows).

I just tried Edge and Chrome, they both opened yesterday too. And they have never opened the website before so it's not a caching issue. 

Timezone, time and date are set correctly on the pi that Birdnet-go is running on.

tomlut, 2024-04-07:
Was this fixed in 0.5.1?

Can't see anything in the release notes but it no longer seems to be happening.

tphakala, 2024-04-07:
No there was no fixes for it in 0.5.1, it is a bug that is happening occasionally.

tphakala, 2024-06-11:
Fixed by https://github.com/tphakala/birdnet-go/pull/144

closed 2024-06-11 by tphakala