tphakala/birdnet-go #114: BirdWeather stops BirdNET-go

MueJosh, 2024-04-08:
Only issue with Birdweather uploads is that uploaded wav files are silent, there is no way that problem covered by this issue would cause loss of any files such as birdnet-go config file.

Please file a new issue with as much details as you can provide.

_Ursprünglich gepostet von @tphakala in https://github.com/tphakala/birdnet-go/issues/68#issuecomment-2041906170_

If I fill the BirdWeather Id and enable BirdWeather upload, the system stops after some time. 
The BirdWeather Id is fine, it works on birdnetpi (I haven't used it for a week or so)
On birdnetgo- BirdWeather doesn't show anything (yep, I waited for an hour)

Website config (set it up on website)
[image]

No output for a few minutes- restarting the console and logging back in and it works again for a few minutes. 
[image]
[image]

Config.yaml
[image]

Some of the output of webui.log

```

[2024-04-08T18:58:45+02:00] [INFO] 172.17.0.1 GET /assets/htmx.min.js 200 <nil>
[2024-04-08T19:05:52+02:00] [INFO] ⇨ http server started on [::]:8080
[2024-04-08T19:07:29+02:00] [INFO] ⇨ http server started on [::]:8080
[2024-04-08T19:08:02+02:00] [INFO] 172.17.0.1 GET / 200 <nil>
[2024-04-08T19:08:03+02:00] [INFO] 192.168.178.23 GET / 200 <nil>
[2024-04-08T19:08:03+02:00] [INFO] 192.168.178.23 GET /assets/tailwind.css 200 <nil>
[2024-04-08T19:08:03+02:00] [INFO] 192.168.178.23 GET /assets/htmx.min.js 200 <nil>
[2024-04-08T19:08:03+02:00] [INFO] 192.168.178.23 GET /assets/custom.css 200 <nil>
[2024-04-08T19:08:03+02:00] [INFO] 192.168.178.23 GET /assets/alpinejs.min.js 200 <nil>
[2024-04-08T19:08:03+02:00] [INFO] 192.168.178.23 GET /top-birds?date= 200 <nil>
[2024-04-08T19:08:03+02:00] [INFO] 192.168.178.23 GET /last-detections?numDetections=5 200 <nil>
[2024-04-08T19:08:04+02:00] [INFO] 192.168.178.23 GET /spectrogram?clip=clips%2F2024%2F04%2Fpasser_domesticus_70p_20240408T185844Z.wav 500 code=500, message=Failed to generate spectrogram
[2024-04-08T19:08:04+02:00] [INFO] 192.168.178.23 GET /spectrogram?clip=clips%2F2024%2F04%2Fstreptopelia_decaocto_74p_20240408T184129Z.wav 200 <nil>
[2024-04-08T19:08:05+02:00] [INFO] 192.168.178.23 GET /spectrogram?clip=clips%2F2024%2F04%2Fstreptopelia_decaocto_79p_20240408T184034Z.wav 200 <nil>
[2024-04-08T19:08:06+02:00] [INFO] 192.168.178.23 GET /spectrogram?clip=clips%2F2024%2F04%2Fstreptopelia_decaocto_72p_20240408T184455Z.wav 200 <nil>
[2024-04-08T19:08:07+02:00] [INFO] 192.168.178.23 GET /spectrogram?clip=clips%2F2024%2F04%2Fstreptopelia_decaocto_74p_20240408T184053Z.wav 200 <nil>
[2024-04-08T19:10:24+02:00] [INFO] ⇨ http server started on [::]:8080
[2024-04-08T19:10:42+02:00] [INFO] 172.17.0.1 GET /settings 200 <nil>
[2024-04-08T19:10:42+02:00] [INFO] 172.17.0.1 GET /assets/tailwind.css 200 <nil>
[2024-04-08T19:10:43+02:00] [INFO] 172.17.0.1 GET /assets/custom.css 200 <nil>
[2024-04-08T19:10:43+02:00] [INFO] 192.168.178.23 GET /assets/htmx.min.js 200 <nil>
[2024-04-08T19:10:43+02:00] [INFO] 192.168.178.23 GET /assets/alpinejs.min.js 200 <nil>
[2024-04-08T19:10:54+02:00] [INFO] 172.17.0.1 POST /update-settings 302 <nil>
[2024-04-08T19:10:54+02:00] [INFO] 172.17.0.1 GET /settings?update=success 200 <nil>
[2024-04-08T19:10:54+02:00] [INFO] 172.17.0.1 GET /assets/tailwind.css 200 <nil>
[2024-04-08T19:10:54+02:00] [INFO] 192.168.178.23 GET /assets/htmx.min.js 200 <nil>
[2024-04-08T19:10:54+02:00] [INFO] 192.168.178.23 GET /assets/alpinejs.min.js 200 <nil>
[2024-04-08T19:10:54+02:00] [INFO] 172.17.0.1 GET /assets/custom.css 200 <nil>
[2024-04-08T19:10:55+02:00] [INFO] 172.17.0.1 GET / 200 <nil>
[2024-04-08T19:10:55+02:00] [INFO] 192.168.178.23 GET /assets/htmx.min.js 200 <nil>
[2024-04-08T19:10:55+02:00] [INFO] 192.168.178.23 GET /assets/alpinejs.min.js 200 <nil>
[2024-04-08T19:10:55+02:00] [INFO] 172.17.0.1 GET /assets/tailwind.css 200 <nil>
[2024-04-08T19:10:55+02:00] [INFO] 172.17.0.1 GET /assets/custom.css 200 <nil>
[2024-04-08T19:10:56+02:00] [INFO] 172.17.0.1 GET /top-birds?date= 200 <nil>
[2024-04-08T19:10:56+02:00] [INFO] 172.17.0.1 GET /last-detections?numDetections=5 200 <nil>
[2024-04-08T19:10:58+02:00] [INFO] 172.17.0.1 GET /spectrogram?clip=clips%2F2024%2F04%2Fstreptopelia_decaocto_72p_20240408T191038Z.wav 200 <nil>
[2024-04-08T19:11:00+02:00] [INFO] 172.17.0.1 GET /spectrogram?clip=clips%2F2024%2F04%2Fturdus_merula_91p_20240408T190805Z.wav 200 <nil>
[2024-04-08T19:11:00+02:00] [INFO] 192.168.178.23 GET /spectrogram?clip=clips%2F2024%2F04%2Fstreptopelia_decaocto_74p_20240408T184129Z.wav 200 <nil>
[2024-04-08T19:11:04+02:00] [INFO] 192.168.178.23 GET /spectrogram?clip=clips%2F2024%2F04%2Fpasser_domesticus_70p_20240408T185844Z.wav 500 code=500, message=Failed to generate spectrogram
[2024-04-08T19:11:16+02:00] [INFO] 192.168.178.23 GET /spectrogram?clip=clips%2F2024%2F04%2Fstreptopelia_decaocto_72p_20240408T184455Z.wav 200 <nil>
[2024-04-08T19:14:38+02:00] [INFO] 172.17.0.1 GET / 200 <nil>
[2024-04-08T19:14:38+02:00] [INFO] 172.17.0.1 GET /assets/tailwind.css 200 <nil>
[2024-04-08T19:14:38+02:00] [INFO] 172.17.0.1 GET /assets/alpinejs.min.js 200 <nil>
[2024-04-08T19:14:38+02:00] [INFO] 172.17.0.1 GET /assets/custom.css 200 <nil>
[2024-04-08T19:14:38+02:00] [INFO] 172.17.0.1 GET /assets/htmx.min.js 200 <nil>
[2024-04-08T19:14:39+02:00] [INFO] 172.17.0.1 GET /last-detections?numDetections=5 200 <nil>
[2024-04-08T19:14:39+02:00] [INFO] 172.17.0.1 GET /top-birds?date=2024-04-08 200 <nil>
[2024-04-08T19:14:44+02:00] [INFO] 172.17.0.1 GET /spectrogram?clip=clips%2F2024%2F04%2Fpasser_domesticus_80p_20240408T191350Z.wav 200 <nil>
[2024-04-08T19:14:45+02:00] [INFO] 172.17.0.1 GET /spectrogram?clip=clips%2F2024%2F04%2Fpasser_domesticus_74p_20240408T191132Z.wav 200 <nil>
[2024-04-08T19:14:45+02:00] [INFO] 172.17.0.1 GET /spectrogram?clip=clips%2F2024%2F04%2Fpasser_domesticus_73p_20240408T191252Z.wav 200 <nil>
[2024-04-08T19:19:18+02:00] [INFO] 192.168.178.23 GET / 200 <nil>

```

tphakala, 2024-05-03:
After almost 6 months of trouble free use I just had this happen to myself, long lasting network connectivity errors to BirdWeather caused birdnet-go process eventually get stuck. I'll fix this soon.

tphakala, 2024-10-16:
I believe this error has been fixed, I have not encountered any birdweather related crashes on two nodes, one running linux and other running windows

closed 2024-10-16 by tphakala