tphakala/birdnet-go #115: LocationFilterThreshold

MueJosh, 2024-04-08:
> Implemented with commit 954c681

 ~~EDIT: "I've just seen that maybe the issue is on my side, I'll get to this in the evening."~~

EDIT_2: It's working flawlessly! Great work and fast commit ❤️ 
Thank you

Great! 

~~But now there is something wrong with the threshold / location~~

#### I have following settings: 
Config.yaml
```
...
# BirdNET model specific settings
birdnet:
    sensitivity: 1.25 # sigmoid sensitivity, 0.1 to 1.5
    threshold: 0.7 # threshold for prediction confidence to report, 0.0 to 1.0
    overlap: 0.5 # overlap between chunks, 0.0 to 2.9
    threads: 4 # 0 to use all available CPU threads
    locale: de # language to use for labels
    latitude: 48.XXXX # latitude of recording location for prediction filtering
    longitude: 7.XXXX # longitude of recording location for prediction filtering
    locationfilterthreshold: 0.0018 # rangefilter species occurrence threshold
...
```

_Ursprünglich gepostet von @MueJosh in https://github.com/tphakala/birdnet-go/issues/96#issuecomment-2033224801_

Unfortunately it still doesn't work as it should.. 
I don't know where the issue could be. 
My settings haven't changed, I updated the container a few days ago. (No idea how I can find out which version I'm running. It says :latest - but it's doing this all the time) 

```

# BirdNET-Go configuration

debug: false # print debug messages, can help with problem solving
# Node specific settings
main:
    name: BirdNET-Go_03 # name of node, can be used to identify source of notes
    timeas24h: true # true for 24-hour time format, false for 12-hour time format
    log:
        enabled: true # true to enable log file
        path: birdnet.log # path to log file
        rotation: daily # daily, weekly or size
        maxsize: 1048576 # max size in bytes for size rotation
        rotationday: 0 # day of the week for weekly rotation, 0 = Sunday
# BirdNET model specific settings
birdnet:
    sensitivity: !!float 1 # sigmoid sensitivity, 0.1 to 1.5
    threshold: 0.3 # threshold for prediction confidence to report, 0.0 to 1.0
    overlap: 0.5 # overlap between chunks, 0.0 to 2.9
    threads: 4 # 0 to use all available CPU threads
    locale: en # language to use for labels
    latitude: 48.XXXX # latitude of recording location for prediction filtering
    longitude: 7.XXX0 # longitude of recording location for prediction filtering
    locationfilterthreshold: 0.0018 # rangefilter species occurrence threshold
# Realtime processing settings

```

Location edited out because privacy, the last digit on longitude is "0" maybe that's an issue? Idk

MueJosh, 2024-04-10:
> I will add more debugging options. Are you able to compile github codebase yourself or do you require pre-compiled binary?
> 
> 
> 
> Compiling is quite easy, just install go 1.22.2, clone github repository and run make

I can try in the next week or so. 
It's the same for every Debian based system? (E.g. raspberry 02w)

tphakala, 2024-04-10:
Yes, compile process is same for any Debian/Ubuntu/Raspberry Pi OS based systems

MueJosh, 2024-04-14:
> Yes, compile process is same for any Debian/Ubuntu/Raspberry Pi OS based systems

Okay, i have it running, but now what. 
I compiled it with make (getting "go" to work was harder than thought) 
Now I'm basically stuck like in my [first discussion](https://github.com/tphakala/birdnet-go/discussions/82)

I can start it with ./birdnet-go realtime but i don't know whats different. 

Since yesterday i have it running in a docker container (docker build with the dockerfile), it works but i think (not tested) the Location Filter Threshold still doesn't do its Thing. 

Should i do something different?

MueJosh, 2024-04-16:
Weird stuff, some birds below 0.01 get detected (LFT set to 0.0006). 
BUT
Not a single Owl (except the Tawny Owl- its everywhere..) gets detected unless I change the Location to 0
then all of them work (including the owls from the species_config.csv - maybe its case sensitive? i have it set up in capital letter [e.g. Eurasian Eagle-Owl] for now) 

This just proves that something is still very wrong with the Location Filter Threshold

MueJosh, 2024-04-16:
the "dog bark filter" also seems to eat up some Owl calls
Im testing it with this video/sounds for now [Youtube European Owls - Sounds](https://youtu.be/hLEXweu1c4c?si=l7EMd19kn7aCqjnJ)
when its set to "false" it detects the eagle owl most of the time- but when its set to "true" its a whole different story... (not an issue at all, I turn it off for now)

tphakala, 2024-04-17:
Dog bark filter is meant to filter out owl calls, at my area almost every single owl detection is a dog. But I am sure there is room for improvement for its logic.

I will try to add some debugging code next weekend to help troubleshoot with location filtering.

MueJosh, 2024-04-17:
> Dog bark filter is meant to filter out owl calls, at my area almost every single owl detection is a dog. But I am sure there is room for improvement for its logic.
> 
> 
> 
> I will try to add some debugging code next weekend to help troubleshoot with location filtering.

Oh, and I thought it was meant to filter out dog barks and not owl calls. Good to know 👍 

Sounds good! Thank you

PeteLawrence, 2024-08-22:
I'm having a problem with detecting Tawny Owls too, I think this may have something to do with the location threshold not being respected.

Running `birdnet-go range print --model latest --threshold 0.0001` or `birdnet-go range print --model legacy --threshold 0.0001` on either the most recent dev build or 0.5.5, I don't see any species with a score below 0.03.

```
root@birdnetgo-6c6c57f4d4-spbjx:/data# birdnet-go range print --model latest --threshold 0.0001
BirdNET-Go build date: 2024-06-09T12:53:30Z, using config file: /root/.config/birdnet-go/config.yaml
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
2024/08/22 09:52:07 Using latest range filter model
Included species for 52.XXXX, 1.XXXX on date 2024-08-22, week 32, threshold 0.000100

Scientific Name                   Common Name                       Score 
--------------------------------- --------------------------------- ------
Columba palumbus                  Common Wood-Pigeon                0.9976
Hirundo rustica                   Barn Swallow                      0.9812
*** LOTS MORE SPECIES ***
Calidris minuta                   Little Stint                      0.0305
Acanthis cabaret                  Lesser Redpoll                    0.0301

Total number of species: 140
```

I notice on https://github.com/tphakala/birdnet-go/blame/31ecd3602a18205bfdca7940243cc905a7182b21/internal/birdnet/rangefilter.go#L142 that 0.03 has been hard-coded, and could possibly be related?  

I've tried to lower this in code and rebuild the binary using the Dev Container, but have not been successful getting the Dev Container running on an M1 MacBook Pro. Initially it complained about tailwindcss and npx missing, which I resolved by updating the build script to install nodejs, but now get an error about the daisyui module missing when trying to run tailwindcss.

PeteLawrence, 2024-08-22:
I've managed to get the code building, and believe I've resolved this in https://github.com/tphakala/birdnet-go/pull/263

tphakala, 2024-10-16:
This is fixed by @PeteLawrence contribution

closed 2024-10-16 by tphakala