tphakala/birdnet-go #120: Feature to override default config file

isZumpo, 2024-04-11:
I'm using kubernetes for running birdnet-go and find it convenient to have all configuration files etc in a separate repository, following the gitops strategy with argocd. One issue I am having currently is the config file. I'm quite happy with the defaults of it and would only like to change certain values. However, right now it is all or nothing, meaning that I have to put it all inside a large configmap, then mount it into the container filesystem.

Would it be possible to add some sort of feature for those who are happy with the default configuration, to override only certain fields of it? For example, lets say I would only want to change webserver port, then maybe a **config-override.yaml** could contain these two fields:

```yaml
webserver:
  port: 80
```

which then would keep the default values, only changing the port to 80 instead of 8080

tphakala, 2024-04-12:
Wouldn't it be more convenient if every setting would have a valid default with or without config file existing? I have been trying to follow this practice but I wouldn't be surprised if I'v missed some. I need to check and fix this.

isZumpo, 2024-04-13:
Yes, that would be even better! From the current code, the config file is created on startup if it does not exist, containing a bunch of predefined defaults. I would imagine this causing some issues if the user were to remove certain fields, or if new fields were added in newer versions of birdnet-go. Since the defaults would not get reapplied, because the config file already exists?

The viper library appears to offers a way of [setting default values](https://github.com/spf13/viper?tab=readme-ov-file#establishing-defaults), that might be better to use compared to writing all default values to a config file on the first startup? Of course, using default values like that would be less visible for the end user... Perhaps the docs could be improved to contain all config values and their defaults?

isZumpo, 2024-05-08:
Fixed in https://github.com/tphakala/birdnet-go/pull/154

closed 2024-05-08 by isZumpo