tphakala/birdnet-go #127: Running birdnet-go in docker on Raspberry Pi

knolljo, 2024-04-20:
I'm currently trying to run birdnet-go on a raspberry pi with docker.

The Pi is running a default, clean Raspbian and docker was installed with this
command `curl -fsSL https://get.docker.com | sh`.

Running this command:
```sh
docker run -ti \
-p 8080:8080 \
--env ALSA_CARD=0 \
--env TZ="Europe/Stockholm" \
--device /dev/snd \
ghcr.io/tphakala/birdnet-go:latest
```

Gives following error message, which suggests that there are no
linux/arm/v7 images for the raspberry pi architecture:
```
Unable to find image 'ghcr.io/tphakala/birdnet-go:latest' locally
latest: Pulling from tphakala/birdnet-go
docker: no matching manifest for linux/arm/v7 in the manifest list entries.
See 'docker run --help'.
```

Adding `linux/arm/v7` to the docker [workflow](https://github.com/tphakala/birdnet-go/blob/main/.github/workflows/docker-build.yml#L71) could maybe build the images for raspberry pi, but I haven't tested yet.

Is there an easier way to run birdnet-go on the pi? I haven't found documentation explicitly for the pi so I'm quite unsure.

Thanks in advance!

isZumpo, 2024-04-20:
Yes, that would make sense to add to the docker building workflow. I suppose that you are running some sort of arm32 based raspberry distro. May I ask exactly what raspberry device model you are using?

knolljo, 2024-04-20:
Of course:
- **OS**: Raspbian GNU/Linux 11 (bullseye) armv7l
- **Host**: Raspberry Pi 3 Model B Plus Rev 1.3
- **Kernel**: 6.1.21-v7+

knolljo, 2024-04-20:
Whops, didn't realize I flashed the 32-bit version, will try again with 64, thanks!

knolljo, 2024-04-20:
Just wanted to add, that I'm doing this for a uni project, where the team
assignment is to do something with KI for birdsong recognition, we are also
backed by [dawn chorus](https://dawn-chorus.org/). Our Prof is pretty open
to suggestions for things to work on, as we are currently in the state of
evaluating projects that we can build upon. We are 6 students in 4-6 semester and
really like birdnet-go for it's simplicity.

I've already skimmed through some feature requests and had some own ideas.
I'm interested what you think about those and if you'd be ok with us contributing some:
- RaspberryPi image with minimal setup, "plug-and-play" distribution
- Microphone Array Support
- Create quizz from detected birds like [birdlingo](https://jessicalieb.itch.io/birdlingo)
- User Documentation

Of course our Prof has to approve this, we are having a meeting on thursday
where I wanted to pitch this idea and having a positive response from you could
help :)

Hope this doesn't sound too weird, thank you.

closed 2024-04-20 by knolljo

tphakala, 2024-04-24:
Yes I am open for contributions.

- Plug and Play distribution is on my todo list, but I think it is too early for it at this point, there are several things missing which I would like to have first, for example easy to use install wizard for web UI for first time run.

- Can you be more specific what microphone array support means? I don't know if it an issue or not but currently BirdNET analyzer can support only one audio channel for analysis

- Quiz sounds a fun feature, if you are able to implement it using Go HTML templates, HTMX and Alpine.JS sure

- Problem with documentation is that I feel that things may still change and evolve a lot, so documentation can get out of date quickly, and for example web ui is still very finished.