tphakala/birdnet-go #130: Features to add from bird weather

matthew73210, 2024-04-28:
Hey,

Just some ideas, they might have already been added or put in the sin bin.
Having a quick look at [Birdweather API](https://app.birdweather.com/api/index.html#introduction), there's some pretty nifty stuff.

As your stated, you mainly want this **_app to be local_**. Maybe a toggle could be added to grab stuff from external sources?

Going back to the API, a nifty thing would be to add the probabilty of said bird to the detections. i.e I sometimes get a false positive of the green sandpiper, I don't think at 1250m in the snow this is all too probable. But there is a 5 percent chance that at my location, one could be seen etc...
One could add weather, or even other sightings.

If it's a nay, pin this so others don't ask.

Cheers,

tphakala, 2024-05-01:
I have been thinking of adding weather data, but I would fetch it from [Openweather](https://openweathermap.org/) instead of BirdWeather. Using BirdWeather for species probabilities is too unreliable I think, there is just far too many false detections with high probability scores.

matthew73210, 2024-05-03:
> I have been thinking of adding weather data, but I would fetch it from [Openweather](https://openweathermap.org/) instead of BirdWeather. Using BirdWeather for species probabilities is too unreliable I think, there is just far too many false detections with high probability scores.

Had a look and I think they just grab the weather stuff from open weather, and predictions from ebird

matthew73210, 2024-05-18:
Been discussing birdweather. But what about ebird? You'll need an API key, but the information from that site is pretty kosher?

tphakala, 2024-05-18:
I have been thinking of adding support for generating [eBird spreadsheet files](https://support.ebird.org/en/support/solutions/articles/48000907878-upload-spreadsheet-data-to-ebird) for reporting bird sighting. This is however pending on UI components for verifying and tagging detections as 100% correct.

matthew73210, 2024-05-18:
This was a discussion from the birdnet-pi erea, of course what you upload has to be validated by oneself. Good idea though

closed 2024-10-16 by tphakala