tphakala/birdnet-go #133: Stopped detecting

farski, 2024-04-29:
Several days ago, my SD card ran out of space so BirdNET mostly stopped working. I cleared out files to make space, but now when I start `birdnet-go` I get some messages that look like errors related to audio capture, which I don't remember showing up before.

```
Read config file: /home/birdnet/.config/birdnet-go/config.yaml
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
Starting BirdNET-Go Analyzer in realtime mode
Threshold: 0.8, sensitivity: 1, interval: 15
[2024-04-29T07:32:15-04:00] [INFO] ⇨ http server started on [::]:8080
Capture Devices
    0: Discard all samples (playback) or generate zero samples (capture), 6e756c6c, [ok]
    1: Playback/recording through the PulseAudio sound server, 64656661756c74, [ok]
    2: Rate Converter Plugin Using Libav/FFmpeg Library, 6c617672617465, [miniaudio: Resource does not exist]
    3: Rate Converter Plugin Using Samplerate Library, 73616d706c6572617465, [miniaudio: Resource does not exist]
    4: Rate Converter Plugin Using Speex Resampler, 737065657872617465, [miniaudio: Resource does not exist]
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
    5: JACK Audio Connection Kit, 6a61636b, [miniaudio: Resource does not exist]
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
    6: Open Sound System, 6f7373, [miniaudio: Resource does not exist]
    7: PulseAudio Sound Server, 70756c7365, [ok]
    8: Plugin using Speex DSP (resample, agc, denoise, echo, dereverb), 7370656578, [miniaudio: Resource does not exist]
    9: Plugin for channel upmix (4,6,8), 75706d6978, [miniaudio: Resource does not exist]
    10: Plugin for channel downmix (stereo) with a simple spacialization, 76646f776e6d6978, [miniaudio: Resource does not exist]
    11: USB Audio Device, USB Audio, 3a312c30, [ok]
Listening ...
```

It had been running perfectly since ~March 30 when I updated to that release. I've updated to the most recent birdnet-go release and updated all the system packages, but I keep getting the same messages. The program is running, and the web UI works, it just never detects anything.

tphakala, 2024-05-01:
Well that's odd .. all I can think of now is that SQLite database would be corrupted somehow but if you say that web UI is working I think it is at least readable?

Can you test if your system is able to record any audio from ALSA device?

Get recording device ID for your system, see (in example below card 0/subdevice 0)
```
# arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Record audio from **hw:(card ID),(subdevice ID)**

```
# arecord -D hw:0,0 -f S16_LE -c1 -r 48000 -d 10 test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
```

If you listen wav file recorded does it contain any sound?

ericleslie, 2024-05-01:
I'm having a problem as well since my SD card has filled. I removed webui logs, spectragraphs and wavs older than 3 days. The docker image starts and the webui loads, but it has stopped detecting birds and there a few spectra graphs it will not generate. Looking at the logs (docker logs -f *image*), it dumps large number of entries and fails saying. 

**Error grabbing logs: invalid character 'l' after object key:value pair**

I assume this is an issue with the last DB entry. I don't have time to look inside and see right this moment.

ericleslie, 2024-05-01:
I found the above error was an error in the docker logs. I deleted the logs and birdnet-go is running normally again. Sorry to hijack this, back to the regularly scheduled program.

tphakala, 2024-05-01:
No worries @ericleslie, glad it is working again for you

farski, 2024-05-03:
This morning I upgraded to the May 1 release, and also finally switched over to a Docker setup, and things are working as expected again

closed 2024-05-03 by farski