tphakala/birdnet-go #138: Primative log support webui.log

matthew73210, 2024-05-03:
Hey, githubs having some issues right now. I can fork, but their backend are down for me so I can't do any pull requests no actully see my fork. Github for vs is having a fit too. So i'll just comment here : 👍 

I've added a little function to read webui.log

_utils.go -->_  
_Add_ import "bufio" 
_Plus the following function_ 

```
// readWebLog reads the content of the web.log file from the root directory and returns it as a string.
// It returns an error if there is an issue opening or reading the file.
func readWebLog() (string, error) {
	// Open the web.log file
	file, err := os.Open("webui.log")
	if err != nil {
		return "", err
	}
	defer file.Close()

	var content string
	scanner := bufio.NewScanner(file)

	// Read the file line by line and append each line to the content string
	for scanner.Scan() {
		content += scanner.Text() + "\n"
	}

	// Check if there was an error during scanning
	if err := scanner.Err(); err != nil {
		return "", err
	}

	// Return the content of the file
	return content, nil
}
```

_In routes.go_
```
// getLogsHandler handles GET requests to the /logs endpoint.
// It reads the content of the webui.log file and renders the logs view with the content.
func (s *Server) getLogsHandler(c echo.Context) error {
	// Read the content of web.log
	logContent, err := readWebLog()
	if err != nil {
		// Return an HTTP error if there is an issue reading the file
		return echo.NewHTTPError(http.StatusInternalServerError, "Failed to read web.log: "+err.Error())
	}

	// Render the logs view and pass the logContent as data
	return c.Render(http.StatusOK, "logs", map[string]interface{}{
		"LogContent": logContent,
	})
}

```
_In the func_ (s *Server) initRoutes() {
_Add the route_ 
```
	// Add a route to handle GET requests to the /logs endpoint
	s.Echo.GET("/logs", s.getLogsHandler)
```

_in logs.html_

```
{{define "logs"}}
<span class="col-span-12">
    {{if .LogContent}}
        <pre>{{.LogContent}}</pre>
    {{else}}
        It didn't work
    {{end}}
</span>
{{end}}
```

It's basic but it does show whats being appended to webui.log (used cat webui.log and i'm getting the same stuff)
Sorry to do it like this but i'll probs forget to do a pull request if i don't do it now. The logs.html needs some work. I did get help from gpt, been ages since i've programmed any webstuff. More used to doing sience stuff (load flow)

tphakala, 2024-05-03:
Thank you! All contributions are welcome no matter how. I will check this during weekend.

matthew73210, 2024-05-07:
I added the small function to utils, but maybe it should go in fileserver?

matthew73210, 2024-05-08:
#153

closed 2024-05-08 by matthew73210