tphakala/birdnet-go #140: Listening device mismatch in Docker

farski, 2024-05-03:
When I run `arecord -l`, I get:

```
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #14 
```

However, when I start the Docker container, I see:

```
BirdNET-Go build date: 2024-05-01T17:38:41Z, using config file: /root/.config/birdnet-go/config.yaml
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
System details: linux debian 12.5 on  hardware
Starting analyzer in realtime mode. Threshold: 0.8, sensitivity: 1, interval: 15
[2024-05-03T08:05:27-04:00] [INFO] ⇨ http server started on [::]:8080
Capture Devices
    0: Discard all samples (playback) or generate zero samples (capture), 6e756c6c, 
    1: Default Audio Device, 64656661756c74, [default]
    2: Default Audio Device, 73797364656661756c74, 
    3: USB Audio Device, USB Audio, 3a312c30, 
Listening on device: Default Audio Device (device ID 64656661756c74)
```

This could just be misunderstanding what the logs are trying to say, or how devices on the host exist with the Docker environment, but I would expect it to be listening on the `USB Audio` device that matches that `arecord` output.

tphakala, 2024-05-03:
```
Capture Devices
    0: Discard all samples (playback) or generate zero samples (capture), 6e756c6c, 
    1: Default Audio Device, 64656661756c74, [default]
    2: Default Audio Device, 73797364656661756c74, 
    3: USB Audio Device, USB Audio, 3a312c30, 
Listening on device: Default Audio Device (device ID 64656661756c74)
```

"Default Audio Device" is "USB Audio Device" in your case, as you do not have any other audio capture devices on your system.

> It is recording clips and when I play them back they sound find, but the detections are not correct; it's not really detecting anything. There are occasional detections, but they are all false positives, it's not picking up any of the normal birds that it would normally be detecting ever couple of minutes.

This I do not understand though .. what hardware are you using? 
How high CPU utilisation your system has when birdnet-go container is running? Could it be that CPU is saturated.

tphakala, 2024-05-03:
On Linux applications typically do not use direct audio hardware devices, there is always some kind of abstraction layer in between. On output you provided your system has Pulse Audio which has control of your USB Audio Device and BirdNET-Go mini audio library has tapped into Pulse Audio server for audio capture, this is expected behavior.

farski, 2024-05-03:
Cool, thanks for the explanation. Running the latest release outside of Docker is yielding more normal results. I can spend some more time on the Docker setup this weekend.

tphakala, 2024-05-10:
Commits 7041fe0bbb33e7ab53b01b2186cb8754abcafe5a and 7f56d6a91b4a8aaf13b45e14c2870af03733b25e introduce user configurable setting for selecting audio source.

config.yaml example
```
realtime:
    audio:
        source: "USB Audio Device"
```

cli example

```
birdnet-go realtime --source "USB Audio Device"
```

closed 2024-06-10 by tphakala