tphakala/birdnet-go #142: Adding some stats to MQTT messages

d0ugal, 2024-05-04:
Hey, I set this up the other day and it's been great so far. Feels way more modern than the others. Great job!

I enabled MQTT so I could do notifications from Home Assistant. However I could do with some more context in the messages.

What do you think about adding number of sightings in a given day to MQTT or some other stats?

I don't really want to be notified for every bird visit but maybe the first time the bird has been seen that day (or that week?).

I'd be interested in helping with this if this would be a welcome change.

tphakala, 2024-05-04:
Hi @d0ugal, thanks for feedback!

I don't personally use MQTT so I would not know how to improve it, if you are willing to work on improvements I would be happy to merge them. Your suggestions sound fine.

d0ugal, 2024-05-05:
Thanks - I'll take a look. I guess more broadly this is about notifications which is why this overlaps with #70.

It would be nice to solve this generally so it can be used beyond MQTT. Then just use it as part of the data payload for MQTT.

tphakala, 2026-02-14:
Closing as resolved. MQTT functionality has been significantly enhanced:
- PR #1749: Home Assistant MQTT auto-discovery with per-source devices, sensors (Last Species, Confidence, Scientific Name, Sound Level)
- PR #1824: Added `detectionId` to MQTT messages for HA URL construction
- PR #1750: Frontend settings UI for HA MQTT auto-discovery
- PR #1763: Sound level template and device name handling

The original request for "number of sightings" and "total species" stats in MQTT is now covered by the HA auto-discovery sensors and the richer MQTT message payload. If additional MQTT fields are needed, please open a new issue.

closed 2026-02-14 by tphakala