tphakala/birdnet-go #159: Suspiciously few detections

farski, 2024-05-10:
As I've been working more closely on Birdnet-Go, I've had some opportunities to be looking at the app while it's running during the day. I've noticed there are times where I'm hearing birds very close to my desk and the microphone I have connected to Birdnet-Go, which should be very clear and easy to detect, but there will be zero detections for a long time. Sometimes 10 minutes will pass with no detections, but there was constant bird activity during that time.

I know it's not apples-to-apples but, just for reference, I ran Merlin for about 45 seconds on my phone, and it had multiple detections of 9 different species. During that same time, Birdnet-Go reported 0 detections. The microphone for Birdnet-Go is in my backyard, and I was just holding Merlin up to the window, so I wouldn't expect it to get better results.

Interestingly, if I play back the clips that are captured for detection from Birdnet-Go and I hold Merlin up to my computer, it will also detect more birds than Birdnet-Go detected for the same clip, so it doesn't seem to be an audio fidelity issue or something like that.

Is there any way I can dig more into what's happening during analysis of chunks, even when it doesn't result in a detection? I see in the code [there](https://github.com/tphakala/birdnet-go/blob/main/internal/birdnet/analyze.go#L62) is [some](https://github.com/tphakala/birdnet-go/blob/main/internal/birdnet/analyze.go#L73) logging that may help with this, but I'm not sure where to see those log statements. They don't show up in the terminal, even when running with the `-d` flag, and I don't actually have a `birdnet.log` file in my Birdnet-Go folder, even though it's enabled in the config file.

tphakala, 2024-05-10:
If you run birdnet-go as a system service log goes to systemd's journal, you see log messages with `journalctl -t birdnet-go` command (this lists full history of birdnet-go log messages).

If you wish to see logs as they come in run `journalctl -t birdnet-go -f`

farski, 2024-05-10:
I got it running as a service based on your [advice](https://github.com/tphakala/birdnet-go/discussions/136) (thanks for that). I'm not seeing a lot of logging related to analysis in the journal output, though

```
ay 10 10:31:01 birdnet birdnet-go[1095]: BirdNET-Go build date: 2024-05-01T17:10:00Z, using config file: /home/birdnet/.config/birdnet-go/config.yaml
May 10 10:31:02 birdnet birdnet-go[1095]: BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
May 10 10:31:02 birdnet birdnet-go[1095]: System details: linux raspbian 12.5 on Raspberry Pi 4 Model B Rev 1.5
May 10 10:31:02 birdnet birdnet-go[1095]:  hardware
May 10 10:31:02 birdnet birdnet-go[1095]: Starting analyzer in realtime mode. Threshold: 0.8, sensitivity: 1, interval: 15
May 10 10:31:03 birdnet birdnet-go[1095]: Logging disabled
May 10 10:31:03 birdnet birdnet-go[1095]: ⇨ http server started on [::]:8080
May 10 10:31:03 birdnet birdnet-go[1095]: Capture Devices
May 10 10:31:03 birdnet birdnet-go[1095]:     0: Discard all samples (playback) or generate zero samples (capture), 6e756c6c,
May 10 10:31:03 birdnet birdnet-go[1095]:     1: Default Audio Device, 64656661756c74, [default]
May 10 10:31:03 birdnet birdnet-go[1095]:     2: Default Audio Device, 73797364656661756c74,
May 10 10:31:03 birdnet birdnet-go[1095]:     3: Rate Converter Plugin Using Libav/FFmpeg Library, 6c617672617465,
May 10 10:31:03 birdnet birdnet-go[1095]:     4: Rate Converter Plugin Using Samplerate Library, 73616d706c6572617465,
May 10 10:31:03 birdnet birdnet-go[1095]:     5: Rate Converter Plugin Using Speex Resampler, 737065657872617465,
May 10 10:31:03 birdnet birdnet-go[1095]:     6: JACK Audio Connection Kit, 6a61636b,
May 10 10:31:03 birdnet birdnet-go[1095]:     7: Open Sound System, 6f7373,
May 10 10:31:03 birdnet birdnet-go[1095]:     8: PulseAudio Sound Server, 70756c7365,
May 10 10:31:03 birdnet birdnet-go[1095]:     9: Plugin using Speex DSP (resample, agc, denoise, echo, dereverb), 7370656578,
May 10 10:31:03 birdnet birdnet-go[1095]:     10: Plugin for channel upmix (4,6,8), 75706d6978,
May 10 10:31:03 birdnet birdnet-go[1095]:     11: Plugin for channel downmix (stereo) with a simple spacialization, 76646f776e6d6978,
May 10 10:31:03 birdnet birdnet-go[1095]:     12: USB Audio Device, USB Audio, 3a302c30,
May 10 10:31:03 birdnet birdnet-go[1095]: Listening on device: Default Audio Device (device ID 64656661756c74)
May 10 10:32:17 birdnet birdnet-go[1169]: BirdNET-Go build date: 2024-05-01T17:10:00Z, using config file: /home/birdnet/.config/birdnet-go/config.yaml
May 10 10:32:17 birdnet birdnet-go[1169]: BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
May 10 10:32:17 birdnet birdnet-go[1169]: System details: linux raspbian 12.5 on Raspberry Pi 4 Model B Rev 1.5
May 10 10:32:17 birdnet birdnet-go[1169]:  hardware
May 10 10:32:17 birdnet birdnet-go[1169]: Starting analyzer in realtime mode. Threshold: 0.8, sensitivity: 1, interval: 15
May 10 10:32:17 birdnet birdnet-go[1169]: 2024/05/10 10:32:17 SQLite database connection initialized: birdnet.db
May 10 10:32:17 birdnet birdnet-go[1169]: Logging disabled
May 10 10:32:17 birdnet birdnet-go[1169]: Initializing context
May 10 10:32:17 birdnet birdnet-go[1169]: Attempting to initialize ALSA backend...
May 10 10:32:17 birdnet birdnet-go[1169]: Loading library: libasound.so.2
May 10 10:32:17 birdnet birdnet-go[1169]: ⇨ http server started on [::]:8080
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_open
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_close
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_sizeof
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_any
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_format
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_format_first
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_get_format_mask
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_channels
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_channels_near
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_channels_minmax
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_rate_resample
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_rate
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_rate_near
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_buffer_size_near
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_periods_near
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_set_access
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_get_format
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_get_channels
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_get_channels_min
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_get_channels_max
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_get_rate
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_get_rate_min
May 10 10:32:17 birdnet birdnet-go[1169]: Loading symbol: snd_pcm_hw_params_get_rate_max
[... cut ...]
May 10 10:40:17 birdnet birdnet-go[1169]: [305.908ms] [rows:1] SELECT n.ID, n.scientific_name, n.clip_name, sub.num_recordings FROM (SELECT * FROM `notes`) AS n INNER JOIN (SELECT ID, scientific_name, ROW_NUMBER() OVER (PARTITION BY scientific_name) as num_recordings FROM `notes` WHERE clip_name != '') AS sub ON n.ID = sub.ID WHERE strftime('%s', 'now') - strftime('%s', begin_time) > 43200 AND sub.num_recordings > 5
May 10 10:40:17 birdnet birdnet-go[1169]: 2024/05/10 10:40:17 Found 1 clips to remove
May 10 10:40:17 birdnet birdnet-go[1169]: 2024/05/10 10:40:17 Removed clips/2024/05/pandion_haliaetus_88p_20240509T223928Z.wav
```

matthew73210, 2024-05-14:
I've sometimes had the analysis part of birdnet-go hang, could this be related? I was getting the same logs as you. Randomly does it just after mid night but restarts after a restart.
Is there a checking mechanism that checks if the recordings are being analysed? Could through an error / restart the container?