tphakala/birdnet-go #164: Guidance for installing on a Mac?

ronbarr, 2024-05-13:
I have an ARM Mac Mini I use as a backup server. I’d also like to add Birdnet-Go. There are no pre-built releases and I tried to build locally using instructions from the Installation section but the build failed with an “Unsupported architecture” error. The docker instructions say it’s Linux only.

I’d love to get this up and running but I don’t see a path forward.

Thanks

tphakala, 2024-05-13:
Hi, thanks for reaching out. I don't have binaries for mac available for now because of Apple's requirement for binaries to be signed, that is a hurdle I have not yet taken.

However compiling should be easy, clone git repo and run make darwin_arm64. I haven't tested this on mac in a long time though.

isZumpo, 2024-05-13:
Please give the docker approach a go. I don't have a apple M1 device to test with personally but I think it should be able to use the arm64 based images which are being produced. The linux only note was added because there were issues with getting mics working on windows based machines.

I'm quite sure the mic issue will also be there for apple devices, however it could be worth trying it anyways. I believe a work around could be to make use of the rtsp feature which birdnet-go offers. Where you could instead listen to a rtsp stream which you start with some mac native program?

tphakala, 2024-05-13:
Current docker arm images are for Linux only

isZumpo, 2024-05-13:
> Current docker arm images are for Linux only

From my understanding **docker run** on a M1 mac should default to the **linux/amd64** version of a given image. Under the hood, docker on a mac [appears ](https://collabnix.com/how-docker-for-mac-works-under-the-hood/) to be working similarly to how it does on windows using wsl. Running inside a linux hypervisor. Of course it will still probably not work as expected given how audio devices work. Possibly some kernel dependencies also? Though, we don't know before someone actually tries it :)

I'm not even sure if it is possible to make a real darwin docker image because it is running inside a linux hypervisor anyways? I see that the darwin_amd64 make target includes some darwin specific tensorflow binaries. However, inside the docker on a mac you would still need the linux ones since it is really running linux?

matthew73210, 2024-05-14:
Docker runs fine for me (M1 mac), just git clone the repo, then "docker build ." in the repo and you are off to the races. The only issue is audio (i use rtsp) it'll fail if you use the alsa route. Doesn't exist on mac. FYI i use it for testing, it builds and runs fine. Didn't check the detections though.
Another option is to use UTM and run on a VM, pass the sound card through and spin up linux.

farski, 2024-05-14:
Also running fine on Mac here using Docker, without detections working

I have a `docker-compose.yml` file that looks like this:

```yml
version: "3.1"
services:
  realtime-analysis:
    build:
      dockerfile: Dockerfile
    ports:
      - "8080:8080"
    volumes:
      - ./birdnet-go-dev/config:/config
      - ./birdnet-go-dev/data:/data
    command: realtime --debug --rtsp rtp://0.0.0.0:9898
```

Running `docker-compose up` should get birdnet-go running, and available at http://localhost:8080 on your Mac.

Note that you don't actually need an RTP stream running at that address for the Docker container to come up and have the web server working.

You can use FFmpeg to create an RTP stream like this: `ffmpeg -f avfoundation -i ":1" -acodec libmp3lame -ab 32k -ac 1 -f rtp rtp://0.0.0.0:9898`. I've had less luck trying to create a `pcm_s16le` stream, which would match what birdnet-go is expecting. The MP3 stream definitely works (I can connect to it using VLC and hear the audio from my Mac's microphone), but I never see any detections when birdnet-go is listening on that stream.

Like @matthew73210 I use this for development, so I don't really need detections, but if you're looking to use this sort of setup as an actual detector, that should get you most of the way there. Some work to do to get the app happier using the stream for detections, though.

tphakala, 2025-01-24:
I'v updated release workflows to build macos (darwin) builds as well, they are included with nightly releases with official releases from v0.6.1 version onwards. https://github.com/tphakala/birdnet-go/releases/tag/nightly-20250124

tphakala, 2026-02-14:
Closing as resolved. macOS (darwin) builds are now included in nightly and official releases (from v0.6.1 onwards), as noted in the last comment on this issue.