tphakala/birdnet-go #173: Pulseaudio 100% CPU

Shellfishgene, 2024-05-22:
Hi!
I'm currently running birdnet-go on a Debian VM under Proxmox. Nothing apart from birdnet-go is running or installed. After a week or so the CPU goes to 100% usage and the VM barely responds anymore. The culprit seems to be pulseaudio, alternating /bin/pulseaudio and /bin/speech-dispatcher use all available CPU. Birdnet-go shows an error 'ring buffer full'. (And stuff about slow database, which is to be expected I guess.)
Is this a Birdnet issue or something with Pulseaudio/Debian?

tphakala, 2024-05-23:
Difficult to say without more details, however 'ring buffer full' would indicate that something failed within BirdNET-Go.

I just released a new version, can you please update to that and use ALSA source instead of PulseAudio source? ALSA is recommeded audio source for time being, you can just uninstall pulseaudio unless you are using it for anything else.

Shellfishgene, 2024-05-23:
Thanks, will try the new version.

Shellfishgene, 2024-07-26:
I removed PulseAudio now, and updated to the current release 0.5.5, and run realtime with -j 1, but the CPU goes to 100% right away with only birdnet-go processes using any. It's a Intel Celeron G4900, one CPU core from Proxmox in a Debian VM. While that's not the greatest CPU it should suffice no?

isZumpo, 2024-07-26:
Maybe you end up with lots of expensive context switching using a single cpu core? Since birdnet has to share cpu time with all other os activities. Even more since it is all running inside a vm. Can you try increasing the cpu core count to two as an experiment?

Shellfishgene, 2024-07-26:
I did that now, after I restart the VM the problem always goes away and only returns after a few days or so. So for now it's back at ~8% CPU. We'll see how it goes.

nicholasmparker, 2024-09-24:
FWIW im getting a ton of "ringbuffer is full" in the logs.  latest version via docker

tphakala, 2024-09-24:
@nicholasmparker can you please provide

- Logs containing "ringbuffer is full" errors, they also contain debug output for this specific issue
- config.yaml, remove any sensitive settings like usernames, passwords, birdweather token, etc
- Detailed description of hardware and OS you are running BirdNET-Go on

Thank you

tphakala, 2024-09-27:
@nicholasmparker can you please test with latest docker image? I have increased ring buffer size and added some additional logging in case error happens. Problem is that I am unable to reproduce this issue myself.

nicholasmparker, 2024-09-27:
Running it now.  I will let you know!

tphakala, 2024-10-16:
Closing this issue for now. If any ring buffer-related issues still occur, I encourage you to open a new issue

closed 2024-10-16 by tphakala