tphakala/birdnet-go #184: Privacy filter too many false positives

tomlut, 2024-06-01:
I was just outside listening to the birds when I did not recognise one, so I looked up my birdnet-go web page to identify it only to see nothing detected in the last hour despite there being plenty of birds easily heard by the microphone. 

Inspecting the journal revealed all my detections were being rejected as containing human voices:

```
Jun 01 16:37:12 raspberrypi birdnet-go[4053]: 2024/06/01 16:37:12 New detection: Gray Butcherbird with confidence: 0.998978
Jun 01 16:37:14 raspberrypi birdnet-go[4053]: 2024/06/01 16:37:14 Updating detection: Gray Butcherbird with confidence: 0.999140
Jun 01 16:37:21 raspberrypi birdnet-go[4053]: 2024/06/01 16:37:21 No species config found, using default actions for gray butcherbird
Jun 01 16:37:24 raspberrypi birdnet-go[4053]: 2024/06/01 16:37:24 Human detected, updating last detection timestamp for privacy filtering
Jun 01 16:37:33 raspberrypi birdnet-go[4053]: 2024/06/01 16:37:33 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 16:37:35 raspberrypi birdnet-go[4053]: 2024/06/01 16:37:35 Human detected, updating last detection timestamp for privacy filtering
Jun 01 16:37:40 raspberrypi birdnet-go[4053]: 2024/06/01 16:37:40 Error retrieving clips for removal: invalid parameters: minHours and minClips must be greater than 0
Jun 01 16:37:42 raspberrypi birdnet-go[4053]: 2024/06/01 16:37:42 Human detected, updating last detection timestamp for privacy filtering
Jun 01 16:37:50 raspberrypi birdnet-go[4053]: 2024/06/01 16:37:50 Human detected, updating last detection timestamp for privacy filtering
Jun 01 16:38:23 raspberrypi birdnet-go[4053]: 2024/06/01 16:38:23 Human detected, updating last detection timestamp for privacy filtering
```

I'm not sure what is being detected as human voiced but it is not actually human voices. 

Is there a sensitivity setting for this?

tphakala, 2024-06-01:
I added sensitivity threshold setting for v0.5.4 release

```
  privacyfilter:          # Privacy filter prevents audio clip saving if human voice 
    enabled: true         # is detected durin audio capture
    threshold: 0.05       # threshold for human voice detection
```

Default value is 0.05, if it is too sensitive set it to lower value, maybe 0.01 or even lower.

tomlut, 2024-06-01:
Hmm. 

I tried as low as 0.00001  and it was still constantly detecting phantom humans.

The dog filter is going off with no dog barking too. 

I have disabled both. 

Even with the dog bark filter disabled I am seeing constant dog detections:

```
  privacyfilter:          # Privacy filter prevents audio clip saving if human voice 
    enabled: false         # is detected durin audio capture
    threshold: 0.00001       # threshold for human voice detection

  dogbarkfilter:
    enabled: false
    condidence: 0.5       # confidence threshold for dog bark detection
    remember: 5           # number of minutes to remember dog barks
```

```
tom@raspberrypi:~ $ journalctl -t birdnet-go -f
Jun 01 19:53:45 raspberrypi birdnet-go[596773]: 2024/06/01 19:53:45 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 19:53:47 raspberrypi birdnet-go[596773]: 2024/06/01 19:53:47 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 19:54:12 raspberrypi birdnet-go[596773]: 2024/06/01 19:54:12 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 19:54:14 raspberrypi birdnet-go[596773]: 2024/06/01 19:54:14 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 19:54:15 raspberrypi birdnet-go[596773]: 2024/06/01 19:54:15 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 19:54:17 raspberrypi birdnet-go[596773]: 2024/06/01 19:54:17 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 19:54:18 raspberrypi birdnet-go[596773]: 2024/06/01 19:54:18 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 19:54:20 raspberrypi birdnet-go[596773]: 2024/06/01 19:54:20 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 19:54:51 raspberrypi birdnet-go[596773]: 2024/06/01 19:54:51 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 19:54:57 raspberrypi birdnet-go[596773]: 2024/06/01 19:54:57 Dog detected, updating last detection timestamp for potential owl false positives
```

tphakala, 2024-06-01:
What is your audio source, is it a RTSP from security camera or a microphone on a sound card?

tomlut, 2024-06-01:
Microphone plugged into a sound card. 

I tried changing back to default in case the high pass filter was creating artefacts. No change. 

Also does not explain why dogs were still being detected when the filter was disabled. (Yes I did save the config and restart birdnet).

tphakala, 2024-06-01:
It seems that I messed up default settings for human privacy and dog bark filters, please add confidence setting for dogbark filter with value of 0.1 or lower and fix a typo for confidence on privacyfilter.

```
  privacyfilter:          # Privacy filter prevents audio clip saving if human voice 
    enabled: true         # is detected durin audio capture
    confidence: 0.05      # threshold for human voice detection

  dogbarkfilter:
    enabled: true
    confidence: 0.1       # confidence threshold for dog bark detection
    remember: 5           # number of minutes to remember dog barks

```

tomlut, 2024-06-01:
Ha! I should have noticed that. 

Seems to have fixed both issues. Thank you.

tomlut, 2024-06-01:
Spoke too soon. It started up again. 
```
Jun 01 21:41:52 raspberrypi birdnet-go[7041]: 2024/06/01 21:41:52 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:42:14 raspberrypi birdnet-go[7041]: 2024/06/01 21:42:14 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:46:09 raspberrypi birdnet-go[7041]: 2024/06/01 21:46:09 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:47:45 raspberrypi birdnet-go[7041]: 2024/06/01 21:47:45 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:48:57 raspberrypi birdnet-go[7041]: 2024/06/01 21:48:57 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:48:58 raspberrypi birdnet-go[7041]: 2024/06/01 21:48:58 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:52:12 raspberrypi birdnet-go[7041]: 2024/06/01 21:52:12 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:54:30 raspberrypi birdnet-go[7041]: 2024/06/01 21:54:30 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:54:46 raspberrypi birdnet-go[7041]: 2024/06/01 21:54:46 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 21:55:39 raspberrypi birdnet-go[7041]: 2024/06/01 21:55:39 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:57:07 raspberrypi birdnet-go[7041]: 2024/06/01 21:57:07 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:57:07 raspberrypi birdnet-go[7041]: 2024/06/01 21:57:07 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 21:57:09 raspberrypi birdnet-go[7041]: 2024/06/01 21:57:09 Human detected, updating last detection timestamp for privacy filtering
Jun 01 21:57:09 raspberrypi birdnet-go[7041]: 2024/06/01 21:57:09 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 21:58:19 raspberrypi birdnet-go[7041]: 2024/06/01 21:58:19 Dog detected, updating last detection timestamp for potential owl false positives
Jun 01 21:58:19 raspberrypi birdnet-go[7041]: 2024/06/01 21:58:19 Human detected, updating last detection timestamp for privacy filtering
Jun 01 22:01:57 raspberrypi birdnet-go[7041]: 2024/06/01 22:01:57 Human detected, updating last detection timestamp for privacy filtering
Jun 01 22:02:49 raspberrypi birdnet-go[7041]: 2024/06/01 22:02:49 Human detected, updating last detection timestamp for privacy filtering
Jun 01 22:05:09 raspberrypi birdnet-go[7041]: 2024/06/01 22:05:09 Human detected, updating last detection timestamp for privacy filtering
Jun 01 22:07:40 raspberrypi birdnet-go[7041]: 2024/06/01 22:07:40 Human detected, updating last detection timestamp for privacy filtering
```

I have disabled both again. This time I think it has stopped. 

I recorded a short audio clip and do not hear any noise that could be causing this. Attached

[test.zip](https://github.com/user-attachments/files/15521750/test.zip)

tomlut, 2024-06-01:
I use this microphone: <https://micbooster.com/clippy-and-pluggy-microphones/65-clippy-em272-microphone.html>

I have just moved the Pi's power plug-pack out of the waterproof enclosure I house the Pi in. The 50Hz AC hum is now below the noise floor - at least during the day when there is a lot of traffic noise, I will have to test again tonight when it is quieter. See (hear) attached. Note that this file has been normalised to a peak amplitude of -1dB in Audacity to hear as much as possible. 

[test_norm.zip](https://github.com/user-attachments/files/15523678/test_norm.zip)

This does appear to have solved the issue. The only dog detections I see now are when there are actually dogs barking and the human detections are likewise operating correctly.

closed 2024-06-01 by tomlut