tphakala/birdnet-go #197: v0.5.4 docker build issue

dskaplan, 2024-06-03:
Hi,

When attempting to pull and build the image tagged as latest (0.5.4). An error or `exec /usr/bin/birdnet-go: exec format error` is returned. This was attempted with the "simple setup". 

I tried a few different set-ups. All of the following were fresh installs and used the latest version of docker. 

* RPi 3b+ with PiOS x64 (bookworm)
* RPi 3b+ with PiOS x64 (bullseye)
* RPi 4 with PiOS x64 (bookworm)
* RPi 4 with PiOS x64 (bullseye)

When I remove the latest tag and specify version 0.5.3, the image builds as expected. Thank  you.

isZumpo, 2024-06-03:
Hi,

There are some issues with the arm64 docker image for this release, see this discussion for more info about it: https://github.com/tphakala/birdnet-go/discussions/185#discussioncomment-9640573

Either wait until next release or use the :dev version of the docker image in the mean time. Sorry for the inconvenience.

tphakala, 2024-06-03:
:dev image should be pretty safe to use, but I have planned to roll out new release this week anyway

dskaplan, 2024-06-03:
No worries, I appreciate the quick reply. Love the project, and I don't have any Go experience, so wanted to add something by testing. Thanks so much.

tphakala, 2024-06-10:
This should be resolved now

closed 2024-06-10 by tphakala