tphakala/birdnet-go #212: miniaudio fails to initialize

hoover67, 2024-06-11:
I just tried v 0.5.5 (thanks for the update!) on my pi4. Sadly, the new  build fails to start up properly while initializing the audio subsystem: 

```
./birdnet-go realtime --latitude 52.11 --longitude 8.7
BirdNET-Go build date: 2024-06-09T13:25:31Z, using config file: /home/pi/.config/birdnet-go/config.yaml
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
2024/06/11 09:18:32 Using latest range filter model
System details: linux raspbian 11.8 on Raspberry Pi 400 Rev 1.1 hardware
Starting analyzer in realtime mode. Threshold: 0.8, overlap: 0, sensitivity: 1, interval: 15
2024/06/11 09:18:33 Clip retention policy: usage
[2024-06-11T09:18:33+02:00] [INFO] ⇨ http server started on [::]:8081
Available Capture Sources:
  0: Discard all samples (playback) or generate zero samples (capture), null
  1: Playback/recording through the PulseAudio sound server, default
  2: Default Audio Device, sysdefault
  3: Rate Converter Plugin Using Libav/FFmpeg Library, lavrate
  4: Rate Converter Plugin Using Samplerate Library, samplerate
  5: Rate Converter Plugin Using Speex Resampler, speexrate
  6: JACK Audio Connection Kit, jack
  7: Open Sound System, oss
  8: PulseAudio Sound Server, pulse
  9: Plugin for channel upmix (4,6,8), upmix
  10: Plugin for channel downmix (stereo) with a simple spacialization, vdownmix
  11: USB Audio, USB Audio, :2,0
2024/06/11 09:18:33 Device init failed miniaudio: Resource does not exist
```
The previous build (from February 2024) still starts up fine. 

```
./backup/birdnet-go realtime --latitude 52.11 --longitude 8.7
Read config file: /home/pi/.config/birdnet-go/config.yaml
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
Starting BirdNET-Go Analyzer in realtime mode
Threshold: 0.8, sensitivity: 1, interval: 15
[2024-06-11T09:19:49+02:00] [INFO] ⇨ http server started on [::]:8081
Listening ...
```

Any idea what could be causing this issue and how to fix it would be much appreciated! 

Thanks & all the best, 

Uwe

tphakala, 2024-06-11:
Could you rename your existing config.yaml file to something else, like config.yaml.bak and let new version to create a a default config file? Config file format has evolved a lot between updates and there is not (yet) any automatic upgrade process for config.

hoover67, 2024-06-11:
thanks for the suggestion which I tried just now, but I'm getting the same error: 

```
pi@pi400:~/birdnet_go$ ./birdnet-go realtime --latitude 52.11 --longitude 8.7 
BirdNET-Go build date: 2024-06-09T13:25:31Z, using config file: /home/pi/.config/birdnet-go/config.yaml
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
2024/06/11 13:08:47 Using latest range filter model
System details: linux raspbian 11.8 on Raspberry Pi 400 Rev 1.1 hardware
Starting analyzer in realtime mode. Threshold: 0.8, overlap: 1.5, sensitivity: 1, interval: 15
Logging disabled
2024/06/11 13:08:47 Clip retention policy: usage
⇨ http server started on [::]:8081
Available Capture Sources:
  0: Discard all samples (playback) or generate zero samples (capture), null
  1: Playback/recording through the PulseAudio sound server, default
  2: Default Audio Device, sysdefault
  3: Rate Converter Plugin Using Libav/FFmpeg Library, lavrate
  4: Rate Converter Plugin Using Samplerate Library, samplerate
  5: Rate Converter Plugin Using Speex Resampler, speexrate
  6: JACK Audio Connection Kit, jack
  7: Open Sound System, oss
  8: PulseAudio Sound Server, pulse
  9: Plugin for channel upmix (4,6,8), upmix
  10: Plugin for channel downmix (stereo) with a simple spacialization, vdownmix
  11: USB Audio, USB Audio, :2,0
2024/06/11 13:08:47 Device init failed miniaudio: Resource does not exist
```

I only changed the port in the newly created config file to 8081 because 8080 is already in use on my pi. 

Thanks, Uwe

tphakala, 2024-06-11:
Well that's odd, can you set audio source to "USB Audio" in config.yaml and test if it works?

```
realtime:
    audio:
        source: "USB Audio"
```

If it does not work please add -d flag to command to enable debug mode, it provides more logs and context.

hoover67, 2024-06-11:
Maybe I should have mentioned that I'm running a classic BirdNET-Instance on the same hardware, alongside birdnet go. 

I tried setting "USB Audio" but it failed during startup with "Device or resource busy". 
...
  11: USB Audio, USB Audio, :2,0
ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave
2024/06/11 13:38:19 Device init failed miniaudio: Device or resource busy
...

I then changed the setting to "Pulse" (which is was birdnet pi defaults to I think) and v0.5.5 starts up just fine now! 

Thanks for the help & all the best, 

Uwe

hoover67, 2024-06-11:
Marking this closed. Maybe a default entry for "audio" in the realtime section could help with a list of possible options.

closed 2024-06-11 by hoover67

tphakala, 2024-06-11:
List of possible options is system dependent, they can vary wildly between Linux systems and are also completely different on Windows and mac.

hoover67, 2024-06-11:
Yep, make sense. Maybe a more helpful error message from birdnet-go would help then if initialization of the default device fails?