tphakala/birdnet-go #217: (Minor) missing greek locale  species names

hoover67, 2024-06-17:
I just noted the following translations seems to be missing from the greek birdnames list: 

```
Λαλοτσιροβάκος                     lesser whitethroat
Φανέτο                                      Eurasian Linnet
Καπακλής                                  Gadwall 
```

Maybe those bird names / translations can be added for the next release?

Thanks,

Uwe

tphakala, 2024-06-19:
Are these translations correct?

When I validated translations with ChatGPT 4 and Claude response is

> The Greek bird species translations you've asked about have some inaccuracies:
> 
> Λαλοτσιροβάκος - This translation for the Lesser Whitethroat is correct. "Λαλοτσιροβάκος" accurately matches the name used for the Lesser Whitethroat in Greek.
> 
> Φανέτο - This name does not correspond to the Eurasian Linnet in Greek. The correct Greek name for the Eurasian Linnet is Κοκκινολαίμης.
> 
> Καπακλής - This name is not the correct Greek translation for the Gadwall. The Gadwall is known as Σαρσέλα in Greek.
> 
> For accurate and detailed information on bird species and their Greek names, it's helpful to refer to reliable bird databases like Avibase or specific bird name dictionaries​ ([Luscinia](https://dic.luscinia.se/))​​ ([AviBase](https://avibase.bsc-eoc.org/avibase.jsp?lang=EN))​.

hoover67, 2024-06-19:
Chat-GPT hallucinating again... kokkinolaimis is definitely the European Robin or Erithacus rubecula. 

https://el.wikipedia.org/wiki/%CE%9A%CE%BF%CE%BA%CE%BA%CE%B9%CE%BD%CE%BF%CE%BB%CE%B1%CE%AF%CE%BC%CE%B7%CF%82

Faneto: (article in greek): 

https://odikapoulia.gr/%CF%84%CE%BF-%CE%BE%CE%AD%CF%81%CE%BF%CF%85%CE%BC%CE%B5-%CF%84%CE%BF-%CF%86%CE%B1%CE%BD%CE%AD%CF%84%CE%BF/

Gadwall ("Schnatterente" in German): 

https://el.wikipedia.org/wiki/%CE%9A%CE%B1%CF%80%CE%B1%CE%BA%CE%BB%CE%AE%CF%82

Thanks & all the best, 

Uwe

tphakala, 2024-06-20:
Okay, just thought to check. Labels updated, thank you.

closed 2024-06-21 by tphakala