tphakala/birdnet-go #221: Crash with "fatal error: concurrent map writes"

ScottESanDiego, 2024-06-18:
If I specify 3 RTSP sources, eventually (usually within an hour) BirdNET-go will crash with "fatal error: concurrent map writes". (See attached file)

[birdnetgocrash.txt](https://github.com/user-attachments/files/15892772/birdnetgocrash.txt)

This is running in the Docker container, launched with `docker run -ti -p 8081:8080 ghcr.io/tphakala/birdnet-go:dev realtime --rtsp 'rtsp://jetson:8554/garage_outdoor_east_sub?video=all&audio=all' --rtsp 'rtsp://jetson:8554/garage_outdoor_north_sub?video=all&audio=all' --rtsp 'rtsp://jetson:8554/front_door?video=all&audio=all'`

Note, this is using the default config file, storing clips in the ephemeral Docker storage, etc.  All the RTSP feeds are from `go2rtc`, Docker environment is a Jetson TX2 with JP4.6 (not that it matters) running in MAXP_ARM_CORE (aka, 4x ARMv8 Cortex-A57 cores active, and the 2x "Denver" ARM cores offline).  Obviously, docker container is the 2024-06-09 (I think!) release.

tphakala, 2024-06-19:
Thank you for reporting. It seems that I had missed mutex locks for map which is holding buffers for multiple streams. Fixed now with https://github.com/tphakala/birdnet-go/pull/223, fix is already available in dev container image, and will be included in next releae.

ScottESanDiego, 2024-06-19:
Looks like it's fixed (or at least way harder to repro! ;-) ) with that commit - thanks for being on top of it!

ScottESanDiego, 2024-06-19:
Fixed in latest container.

closed 2024-06-19 by ScottESanDiego