tphakala/birdnet-go #229: Adding BirdWeather API Token crashed my docker container

have-no-clue-what-im-doing, 2024-06-22:
About 20 minutes after adding my BirdWeather API Token to the config (Through the web GUI) my container crashed. When I went to restart the container I got this error:
`Error loading settings: error initializing viper: fatal error reading config file: While parsing config: yaml: cannot decode !!str `my_birdweather_api_token as a !!int`

Did BirdWeather tokens use to be only int based?

I edited the config file and changed `!!int` to `!!str`
Started the container back up and all is running well.

I think the container crashed only when it detected a bird and was trying to upload it to BirdWeather and trying to use the API key. It's working perfectly fine now and is properly uploading data to BirdWeather

tphakala, 2024-06-23:
How did you enter token the first time, through web ui or did you edit config.yaml with editor like nano or vim?

florisre, 2024-06-23:
I am experiencing the same issue. I first entered the api key using the web interface. This inserted it into the ID field in the config file, but with the `!!int` prefix, as mwntioned before.

have-no-clue-what-im-doing, 2024-06-23:
> How did you enter token the first time, through web ui or did you edit config.yaml with editor like nano or vim?

Yeah I updated through the web GUI. I am using the latest non dev docker image.

tphakala, 2024-06-24:
Okay, it is a bug in web ui settings then, I'll fix this for next release.

emilic14, 2024-08-21:
Same issue here ^^

tphakala, 2024-08-21:
Please try with latest dev image

emilic14, 2024-08-21:
I actually just built it and had the issue when using the web ui.
I've rebuilt the image but added the token in the config yaml, and that seems to be working ok (at least no crash on the container)

hoover67, 2024-08-30:
I'm having the same issue with the release from June 9th. Where can I find the latest "dev image"? Thanks! 

Uwe

florisre, 2024-09-01:
> I'm having the same issue with the release from June 9th. Where can I find the latest "dev image"? Thanks!
> 
> Uwe

Try here: https://github.com/tphakala/birdnet-go/pkgs/container/birdnet-go/266615682?tag=dev

tphakala, 2025-01-14:
I believe this issue has been resolved

closed 2025-01-14 by tphakala