tphakala/birdnet-go #230: Missing thumbnail for certain species

xconverge, 2024-06-23:
I am missing a thumbnail for just one species? (so far that I have noticed)

```
Corvus brachyrhynchos
```

Should be this image I believe https://en.wikipedia.org/wiki/American_crow#/media/File:Corvus-brachyrhynchos-001.jpg

I don't see anything that looks suspicious in the logs

isZumpo, 2024-06-23:
Some birds will be excluded since their license is too troublesome to parse into something readable. In other cases the author information might be missing, which appears be the case for this image:

https://en.wikipedia.org/w/api.php?action=query&prop=imageinfo&iiprop=extmetadata&titles=File%3aCorvus-brachyrhynchos-001.jpg&format=json

**Files with no machine-readable author**

Not sure how to deal with these cases right now. Maybe we can work around it somhow or do some sort of author override list for certain birds 🤔

xconverge, 2024-06-24:
https://creativecommons.org/licenses/by-sa/3.0/deed.en#ref-appropriate-credit
https://wiki.creativecommons.org/wiki/License_Versions#Detailed_attribution_comparison_chart

"if supplied"

to me looks like it is fine to still use the image and make the author check optional in this case

isZumpo, 2024-06-24:
Wonderful! Perhaps we could add some logic to handle missing authors for creative commons licenses

tphakala, 2025-01-14:
This issue is now resolved, I relaxed media queries so that also images without copyright information are allowed.

closed 2025-01-14 by tphakala