tphakala/birdnet-go #237: Not receiving MQTT messages + MQTT status indicator feature request

PovilasID, 2024-06-27:
Hey,

I pulled down a new dev build ([5fa713cedf358e5baa49c609c1ba05ade425a1df](https://github.com/users/tphakala/packages/container/birdnet-go/236031669?tag=5fa713cedf358e5baa49c609c1ba05ade425a1df)) but it seams that somewhere along the way  broke MQTT message publishing. 
Log still reports: `Connected to MQTT broker: tcp://localhost:1883`
But I do not see any detections messages in the MQTT explorer or in the previously configured Home Assistant sensor log. 

I could go back a few images and wait for an owl to check on what build MQTT stopped working for but... that is slow debugging process, so I have a suggestion. You can actually save data to MQTT server. For example that is how things like OpenMQTTGateway maintain a 'status' ([LWT](https://docs.openmqttgateway.com/upload/portal.html#configuration-portal) and save some of their settings). Status indicator is present in MQTT even if the device or application is offline. That is probably ideal but any kind of status indicator in MQTT specifying it's status: online/offline/detecting/crashing would be useful even if it is present only while app is running. MQTT can be used to notify about detection but also technical issues.

thor0215, 2024-06-27:
Can you provide more details on your MQTT and Birdnet-Go setup?

tphakala, 2024-06-28:
@PovilasID apologies, my yestedays updates to MQTT action included a snafu which blocked all MQTT updates, fixed with https://github.com/tphakala/birdnet-go/commit/9f9bc15eac4b018aa4a830aedd829a32cf2f3f71

I have been thinking about adding status display in Web UI for MQTT and rest of services, so yes this is planned and likely to be added soon.

PovilasID, 2024-06-28:
@tphakala no problem. It is working fine now! It is `dev` branch I do not expect it to be stable or bug free and I am fine catching some bugs and reporting back 👍 
My idea was a status indicator in MQTT not for MQTT. It is for 3rd party apps, so that they could log when detection is on or off or if there is an error. I assume you are not planning on adding notification functionality any time soon.

@thor0215 it is nothing special just `localhost:port` and a topic. If I find some time I set up a panel in home assistant showing last detected bird and will share the setup. It should be possible all components are there.

closed 2024-06-28 by PovilasID

tphakala, 2024-06-28:
Okay, I need to think about application status delivery over MQTT.

PovilasID, 2024-06-28:
I would again suggest looking at projects that already do that like https://github.com/1technophile/OpenMQTTGateway or https://github.com/blakeblackshear/frigate or https://github.com/Koenkk/zigbee2mqtt they all have it.
MQTTGateway probably has the most developed model out all of them where you can even change settings on devices by pushing MQTT commands.