tphakala/birdnet-go #239: Confidence level Threshold

matthew73210, 2024-06-29:
Hello,

Right now i have my threshold set at 0.8 

```
# BirdNET model specific settings
birdnet:
    sensitivity: !!float 1 # sigmoid sensitivity, 0.1 to 1.5
    threshold: 0.8 # threshold for prediction confidence to report, 0.0 to 1.0
    overlap: 0.0 # overlap between chunks, 0.0 to 2.9
    threads: 0 # 0 to use all available CPU threads
    locale: en # language to use for labels
    latitude: xxx # latitude of recording location for prediction filtering
    longitude: xxx # longitude of recording location for prediction filtering
    rangefilter:
        version: latest # model to use for range filter: "latest" or "legacy" for previous model
        threshold: 0.01 # rangefilter species occurrence threshold
```

Yet i'm getting the following :

[image]

Is the perhaps related to the dynamic threshold, or is this still not implemented in the main repo?

jasonmadigan, 2024-08-30:
Seeing similar, would like to understand better.

tphakala, 2024-08-30:
It is likely due to dynamic threshold which is enabled by default, it is explained (including config parameters) here in more detail https://github.com/tphakala/birdnet-go/discussions/211

Lenbok, 2025-07-08:
@tphakala That's much clearer thanks. It would be good to emit a log message when the dynamic threshold for a species gets adjusted so we can inspect what's happening.

closed 2026-02-14 by tphakala