tphakala/birdnet-go #248: Recordings getting lost - UI clogging up with faulty spectograms

florisre, 2024-07-09:
For some reason, I am getting huge, enpty spectograms for some recognitions. They fill up lots and lots of space on my screen – multiple screen lengths. 

This might be related to constant recognitions of one species together with my config allowing up to three recordings of one species when space gets tight. 

[image]
[image]

tphakala, 2024-07-10:
Which version of application you are using?

florisre, 2024-07-10:
the docker image with the `latest` tag.

tphakala, 2026-02-14:
Closing as stale. This issue is from July 2024 describing UI rendering problems with spectrograms in the old frontend. The UI has been completely rewritten in Svelte, and significant improvements to audio clip management have been made since. If this issue persists on a current build, please open a new issue with updated details.

closed 2026-02-14 by tphakala