tphakala/birdnet-go #251: ringbuffer is full errors

jeroenterheerdt, 2024-07-19:
After a short while (2 minutes or so) I see a lot of this in my log:

```2024/07/18 16:59:30 Error writing to ring buffer for stream rtsp://192.168.1.15/ch0_2.h264: ringbuffer is full```

then it goes away, and will eventually come back. Is there any way for me to increase the ring buffer size?

tphakala, 2024-07-19:
Buffers should not run out if hardware is fast enough to process incoming audio data.
Which version of BirdNET-Go are you using, and on which hardware?
How many audio sources do you have?

If your hardware is limited in performance you can reduce time it takes to process audio by lowering BirdNET overlap setting, good values to try are 0.5 and 0. This reduces detection accuracy slightly.

jeroenterheerdt, 2024-07-19:
Just the one audio source and I have installed birdnet yesterday (so latest version). I am running on a docker install on alpine Linux proxmox vm on a Nuc equivalent.  Physical hardware should not be the problem, but maybe somewhere in the many layers a limit is introduced? I am using an rtsp stream.

PeteLawrence, 2024-08-10:
I too have noticed experienced this problem.

Hardware: Raspberry Pi 5
BirdNET-Go build date: 2024-08-02T12:43:14Z
BirdNET-Go is running via Kubernetes with 1 CPU core and 1GB RAM allocated to it
2 RTSP streams hosted on MediaMTX

When first starting up BirdNet-Go it runs fine using < 40% of the allocated CPU and < 256 MB RAM. After an amount of time however (sometimes almost immediately, sometimes after a week or two) I start getting the ringbuffer is full errors. Logging on to the Raspberry Pi I can see 10+ ffmpeg processes running, each consuming enough resources to cause the BirdNet-Go process to get throttled presumably leading to the ringbuffer errors. So it seems like under some circumstances new ffmpeg processes are being created, but the old ones not being terminated. I'm not sure if this is specific to my setup, or is a wider issue, so sharing here hoping we can track down the root cause.

tphakala, 2024-08-12:
Thanks Pete for detailed description, this helps troubleshooting

PeteLawrence, 2024-08-26:
Thanks very much, I've pulled in the latest dev release and will let you know how it goes.

tphakala, 2024-08-26:
I've implemented an another update that captures system resource usage and the process table (on Linux systems) if a ring buffer write operation fails. This debug data is now written to the log file. This additional information should help in diagnosing the reported issues.

PeteLawrence, 2024-09-08:
The recent changes appear to have significantly improved things for me.

I believe the root cause of the problems was poor wifi signal between one of microphones and the MediaMTX server that I run, causing the audio stream to arrive in bursts/garbled, or not at all, leading to the ffmpeg process restarting, but not necessarily removing the now dead process. Now when this problem occurs the ffmpeg process is restarted correctly, allowing birdnetgo to continue processing.

A couple of times over the past fortnight I've had a microphone stream go offline altogether which would've previously hung up birdnetgo, now however it will continue processing the remaining healthy stream, whilst retrying the broken stream until it comes back online. 

Thanks for sorting this!

closed 2024-09-22 by tphakala