tphakala/birdnet-go #279: Thumbnails not appearing

pack2010, 2024-09-18:
Hello!

I've been getting detections but no thumbnail appears and I get this error message. Not sure what to do with it.

Failed to get or generate spectrogram for clip clips/2024/09/otus_sunia_84p_20240917T195637Z.wav: error creating spectrogram with SoX: SoX binary not found: exec: "sox": executable file not found in %PATH%

Thanks so much in advance!

tphakala, 2024-09-18:
Which version are you using? Thumbnail support isn't in latest binary release 0.5.5 yet, but it is in docker dev version.

Spectrograms are being built by sox (https://github.com/chirlu/sox) so you need to have sox installed to see spectrograms, on raspberri pi os you can install it by "sudo apt install sox"

pack2010, 2024-09-18:
I'm in 0.5.5. Should I go to earlier version? I thought I installed sox but I'll try to use the link you provided.
Also having problems getting to upload to Bird Weather. 

2024/09/18 16:46:05 Failed to upload soundscape to Birdweather: upload failed, response reported failure
2024/09/18 16:46:05 error uploading to BirdWeather: upload failed, response reported failure
2024/09/18 16:46:05 Error executing action: upload failed, response reported failure

tphakala, 2024-09-19:
If you want thumbnail support at this time you have two options:
- Compile BirdNET-Go from source code available in this repository
- Switch to Docker container dev image version which is built every time I update code in this repository

Or wait until I make v0.6.0 release.

Regarding sox github link was just a reference. if you have sox installed is it in system path? If you run "sox" command in Linux shell does it work?

I can't help with Birdweather problem without more details

Do you have a valid Birdweather token from birdweather.com?
Has uploads ever worked?

But given error message you have getting "response reported failure" would indicate that something failed at birdweather.com side, maybe because of incorrect token etc.

pack2010, 2024-09-19:
Ok, I might just wait for 0.6 release, it's not a big deal at all. I tried to install sox but when I try to do the permanent path setup it tells me my syntax is bad. I have a birdweather account, is that the same as a token? I put my user ID in as the input to upload the clips.

tphakala, 2025-01-24:
v0.6.0 is now released

closed 2025-01-24 by tphakala