tphakala/birdnet-go #287: Audio player controls are not working

tphakala, 2024-10-01:
Hi @tphakala! A very happy user here. This new player looks gorgeous. However, it isn't working for me. It does something, and I can see it in the inspector, but something seems broken. It's happening both in Firefox and Chrome (Fedora Linux in both cases).

https://github.com/user-attachments/assets/1ebace4a-f678-4f35-84dc-9b1d9403c41d

_Originally posted by @vibragiel in https://github.com/tphakala/birdnet-go/discussions/285#discussioncomment-10807806_

tphakala, 2024-10-01:
Can you please open a species detection listing and look for errors in developer mode console on that page? Dashboard recent detections is still using old player so any errors with new player are now showing up while on dashboard.

vibragiel, 2024-10-01:
This is the only error that appears just after entering the dashboard:

[image]

There are no new errors when opening a species detection, and the rest of the page seems to work fine otherwise.

vibragiel, 2024-10-01:
FYI, I downgraded to an image from before you introduced the new audio player and the htmx syntax error was there already.

tphakala, 2024-10-01:
Yeah that HTMX error is not nice, but it is not related to this issue as there is no HTMX in use for audio player code, just javascript. Does audio player for recent detections work?

vibragiel, 2024-10-01:
> Does audio player for recent detections work?

Yes, that one works fine.

Tserppa, 2024-10-01:
Hi, I can see this issues as well but the new player works if I scale the window down. Just try to scale the window down so it's as narrow as a mobile phone screen.

tphakala, 2024-10-01:
Which OS are you using @Tserppa Linux?

Tserppa, 2024-10-01:
Server is Linux but I'm viewing the dashboard from Windows 11 + Edge.

tphakala, 2024-10-01:
OK, I see this also when using docker image from github, I guess docker build workflow is broken. I'll see if I can fit it today. Thanks for reporting @vibragiel and @Tserppa

tphakala, 2024-10-01:
Problem is now fixed in image ghcr.io/tphakala/birdnet-go:e96b221de621941fbdcbe2670d450b27802acfe9, I had forgot to commit updated CSS files which broke bunch of things in the UI

Tserppa, 2024-10-01:
I can confirm it working now. And btw this is a great update. Thanks @tphakala!

vibragiel, 2024-10-01:
Working flawlessly. Thanks!

vibragiel, 2024-10-01:
@tphakala Let me know if you need help debugging the htmx syntax error, though it doesn't seem to affect functionality in any way I can see.

tphakala, 2024-10-01:
@vibragiel Thanks, HTMX code is directly from HTMX upstream so bug could be simply resolved by updating HTMX library. But I have been thinking of replacing HTMX code with alpine.js + native javascript, haven't made my mind yet though.

vibragiel, 2024-10-14:
The htmx error is gone now (perhaps thanks to [this](https://github.com/tphakala/birdnet-go/pull/292/commits/6ce550f9f8030a0d00687c9c4fc139dd6cadf110)?), so as far as I'm concerned, this issue can be closed. Thanks!