tphakala/birdnet-go #295: Crash when attempting to upload to BirdWeather

vibragiel, 2024-10-14:
I just enabled the uploads to BirdWeather and got this error on the first detection:

```
birdnet-go_1  | 2024/10/14 16:31:16 New detection: Gorrión Común with confidence: 0.87, source: malgo
birdnet-go_1  | panic: runtime error: invalid memory address or nil pointer dereference
birdnet-go_1  | [signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0xb1346c]
birdnet-go_1  | 
birdnet-go_1  | goroutine 43 [running]:
birdnet-go_1  | github.com/tphakala/birdnet-go/internal/birdweather.(*BwClient).UploadSoundscape(0x0, {0x400156e7a0, 0x1c}, {0x4002422000?, 0x6126f60?, 0xb49b28?})
birdnet-go_1  | 	/home/dev-user/src/BirdNET-Go/internal/birdweather/birdweather.go:108 +0x38c
birdnet-go_1  | github.com/tphakala/birdnet-go/internal/birdweather.(*BwClient).Publish(_, {0x0, {0x4000382640, 0xa}, {0x4000011020, 0xa}, {0x4000011070, 0x8}, {0xf926d8, 0x5}, ...}, ...)
birdnet-go_1  | 	/home/dev-user/src/BirdNET-Go/internal/birdweather/birdweather.go:260 +0x1bc
birdnet-go_1  | github.com/tphakala/birdnet-go/internal/analysis/processor.BirdWeatherAction.Execute({0x40002d0f08, {0x0, {0x4000382640, 0xa}, {0x4000011020, 0xa}, {0x4000011070, 0x8}, {0xf926d8, 0x5}, ...}, ...}, ...)
birdnet-go_1  | 	/home/dev-user/src/BirdNET-Go/internal/analysis/processor/actions.go:174 +0x80
birdnet-go_1  | github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).actionWorker(0x0?)
birdnet-go_1  | 	/home/dev-user/src/BirdNET-Go/internal/analysis/processor/workers.go:43 +0xdc
birdnet-go_1  | created by github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).startWorkerPool in goroutine 1
birdnet-go_1  | 	/home/dev-user/src/BirdNET-Go/internal/analysis/processor/workers.go:34 +0x68
```

Well I'm not sure it's even tried to upload, because the threshold is set at `0.9` and the detection had a `0.87` confidence.

vibragiel, 2024-10-14:
Ok the next detection went well, so I'm not sure what happened earlier.

```
birdnet-go_1  | 2024/10/14 16:40:52 New detection: Estornino Negro with confidence: 0.85, source: malgo
birdnet-go_1  | 2024/10/14 16:41:02 Saved audio clip to clips/2024/10/sturnus_unicolor_85p_20241014T164052Z.m4a
birdnet-go_1  | 2024/10/14 16:41:03 Response Body: {"success":true,"soundscape":{"id":1084053980,"stationId":8201,"timestamp":"2024-10-14T16:40:50.000+00:00","url":null,"filesize":577580,"extension":null,"duration":null,"mode":"birdnetpi"}}
birdnet-go_1  | 2024/10/14 16:41:03 Soundscape successfully posted to Birdweather
birdnet-go_1  | 2024/10/14 16:41:03 Posting detection to Birdweather:  https://app.birdweather.com/api/v1/stations/BUgMwyBgEn2ggX9g9Cr773jd/detections
birdnet-go_1  | 2024/10/14 16:41:03 JSON Payload: {"timestamp":"2024-10-14T16:40:50.000+0000","lat":37.3816,"lon":-6.1379,"soundscapeId":"1084053980","soundscapeStartTime":"2024-10-14T16:40:50.000+0000","soundscapeEndTime":"2024-10-14T16:40:53.000+0000","commonName":"Estornino Negro","scientificName":"Sturnus unicolor","algorithm":"2p4","confidence":"0.85"}
```

I'll keep monitoring.

tphakala, 2024-10-14:
Which version of BirdNET-Go are you using?

tphakala, 2024-10-15:
This commit should make birdweather related code more robust https://github.com/tphakala/birdnet-go/commit/c1433cb2d98a9f1367efdf66743be0c4bcad3abc

vibragiel, 2024-10-15:
Thanks! I'll keep an eye tocheck it doesn't appear anymore.

closed 2024-10-15 by vibragiel