tphakala/birdnet-go #296: Crash when listening to multiple RTSPS streams

jdoss, 2024-10-14:
I just setup BirdNet-Go in a container and I configured it to listen to four Unifi Protect cameras and it was working great until it crashed with the following error. This is with the `latest` container tag.

```
fatal error: concurrent map writes

goroutine 102 [running]:
github.com/tphakala/birdnet-go/internal/myaudio.readFromBuffer({0xc0003eff80, 0x34})
        /root/src/BirdNET-Go/internal/myaudio/buffers.go:91 +0x191
github.com/tphakala/birdnet-go/internal/myaudio.BufferMonitor(0x0?, 0xc0003cd220, 0xc000125da0, {0xc0003eff80, 0x34})
        /root/src/BirdNET-Go/internal/myaudio/buffers.go:117 +0x13e
created by github.com/tphakala/birdnet-go/internal/analysis.RealtimeAnalysis in goroutine 1
        /root/src/BirdNET-Go/internal/analysis/realtime.go:123 +0x7bd

goroutine 1 [select, 23 minutes]:
github.com/tphakala/birdnet-go/internal/analysis.RealtimeAnalysis(0xc000382a88)
        /root/src/BirdNET-Go/internal/analysis/realtime.go:142 +0x9f8
github.com/tphakala/birdnet-go/cmd/realtime.Command.func1(0xc0003b4c08?, {0xf324dd?, 0x4?, 0xf323cd?})
        /root/src/BirdNET-Go/cmd/realtime/realtime.go:20 +0x17
github.com/spf13/cobra.(*Command).execute(0xc0003b4c08, {0x5ffaac0, 0x0, 0x0})
        /go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:983 +0xaca
github.com/spf13/cobra.(*Command).ExecuteC(0xc0003b4308)
        /go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1115 +0x3ff
github.com/spf13/cobra.(*Command).Execute(0xc000382a88?)
        /go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1039 +0x13
main.main()
        /root/src/BirdNET-Go/main.go:39 +0x174

goroutine 69 [select, 23 minutes]:
database/sql.(*DB).connectionOpener(0xc00018f380, {0x10c36e8, 0xc0002a01e0})
        /usr/local/go/src/database/sql/sql.go:1246 +0x87
created by database/sql.OpenDB in goroutine 1
        /usr/local/go/src/database/sql/sql.go:824 +0x14c

goroutine 71 [chan receive]:
github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).startDetectionProcessor.func1()
        /root/src/BirdNET-Go/internal/analysis/processor/processor.go:141 +0x55
created by github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).startDetectionProcessor in goroutine 1
        /root/src/BirdNET-Go/internal/analysis/processor/processor.go:139 +0x4f

goroutine 72 [chan receive, 3 minutes]:
github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).actionWorker(0x0?)
        /root/src/BirdNET-Go/internal/analysis/processor/workers.go:40 +0x6b
created by github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).startWorkerPool in goroutine 1
        /root/src/BirdNET-Go/internal/analysis/processor/workers.go:34 +0x65

goroutine 73 [chan receive, 14 minutes]:
github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).actionWorker(0x0?)
        /root/src/BirdNET-Go/internal/analysis/processor/workers.go:40 +0x6b
created by github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).startWorkerPool in goroutine 1
        /root/src/BirdNET-Go/internal/analysis/processor/workers.go:34 +0x65

goroutine 74 [chan receive, 10 minutes]:
github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).actionWorker(0x0?)
        /root/src/BirdNET-Go/internal/analysis/processor/workers.go:40 +0x6b
created by github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).startWorkerPool in goroutine 1
        /root/src/BirdNET-Go/internal/analysis/processor/workers.go:34 +0x65

goroutine 75 [chan receive, 7 minutes]:
github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).actionWorker(0x0?)
        /root/src/BirdNET-Go/internal/analysis/processor/workers.go:40 +0x6b
created by github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).startWorkerPool in goroutine 1
        /root/src/BirdNET-Go/internal/analysis/processor/workers.go:34 +0x65

goroutine 76 [chan receive, 4 minutes]:
github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).actionWorker(0x0?)
        /root/src/BirdNET-Go/internal/analysis/processor/workers.go:40 +0x6b
created by github.com/tphakala/birdnet-go/internal/analysis/processor.(*Processor).startWorkerPool in goroutine 1
        /root/src/BirdNET-Go/internal/analysis/processor/workers.go:34 +0x65

goroutine 77 [chan receive, 6 minutes]:
[... 276 lines cut ...]
os/exec.(*Cmd).Wait(0xc001f04000)
        /usr/local/go/src/os/exec/exec.go:897 +0x45
github.com/tphakala/birdnet-go/internal/myaudio.captureAudioRTSP.func1()
        /root/src/BirdNET-Go/internal/myaudio/rstp.go:47 +0x25
created by github.com/tphakala/birdnet-go/internal/myaudio.captureAudioRTSP in goroutine 111
        /root/src/BirdNET-Go/internal/myaudio/rstp.go:46 +0x50d

goroutine 148 [IO wait]:
internal/poll.runtime_pollWait(0x7f93478262f0, 0x72)
        /usr/local/go/src/runtime/netpoll.go:345 +0x85
internal/poll.(*pollDesc).wait(0xc001f06060?, 0xc001f24000?, 0x1)
        /usr/local/go/src/internal/poll/fd_poll_runtime.go:84 +0x27
internal/poll.(*pollDesc).waitRead(...)
        /usr/local/go/src/internal/poll/fd_poll_runtime.go:89
internal/poll.(*FD).Read(0xc001f06060, {0xc001f24000, 0xffff, 0xffff})
        /usr/local/go/src/internal/poll/fd_unix.go:164 +0x27a
os.(*File).read(...)
        /usr/local/go/src/os/file_posix.go:29
os.(*File).Read(0xc001f08000, {0xc001f24000?, 0xc001f24000?, 0x780?})
        /usr/local/go/src/os/file.go:118 +0x52
github.com/tphakala/birdnet-go/internal/myaudio.captureAudioRTSP.func2()
        /root/src/BirdNET-Go/internal/myaudio/rstp.go:67 +0x10a
created by github.com/tphakala/birdnet-go/internal/myaudio.captureAudioRTSP in goroutine 111
        /root/src/BirdNET-Go/internal/myaudio/rstp.go:55 +0x5fc
```

tphakala, 2024-10-14:
Thanks for detailed report, looks like there is some concurrency issue caused by multiple streams. I'll look into this tomorrow.

jdoss, 2024-10-14:
No problem! Thanks for your quick response and for all of your hard work on this project. I am very willing to help you with any testing.

tphakala, 2024-10-15:
This is now fixed by https://github.com/tphakala/birdnet-go/commit/88ea0cb6e883ac67f2332347c62d600e51accc64

jdoss, 2024-10-16:
I built a container from source and I have been running it for over 24hr four RTSPS streams without issues. I think this can be closed out. Thanks @tphakala!!

closed 2024-10-16 by jdoss

tphakala, 2024-10-16:
Currently containers in Github registry are built for every commit, so you could have just pulled new image with latest tag. However it is good that you were able to build container by youself 👍