tphakala/birdnet-go #297: Birdweather upload ignores treshold

Mraedis, 2024-10-14:
I noticed that _any_ detection gets uploaded to birdweather, even though the default .80 threshold for uploads is in place.
Don't mind the .79 ones getting through but the .54 are a bit much.

I am running dev @ sha256:b5946036c133fc835a17e2a5de7e3d59e949ef3a332839cb3bdf0dfddca8f9da
Perhaps this has to do with the dynamic tresholds?

tphakala, 2024-10-15:
This is now fixed by https://github.com/tphakala/birdnet-go/commit/558d093a988845c2c5ad3f566bcc2c149005653b

vibragiel, 2024-10-15:
I can cofirm it works now :+1:

Mraedis, 2024-10-16:
I will let you know once I get a lower threshold detection, it's been nothing but 90+ the past day. 👀

Mraedis, 2024-11-07:
Just wanted to note that it's definitely fixed. :)

closed 2024-11-07 by Mraedis