tphakala/birdnet-go #300: Recording browser for UI

tphakala, 2024-10-16:
The Web UI requires an improved interface for browsing detections and recordings beyond the current search function. Users should be able to filter and browse recordings based on specific criteria, including at minimum:

- Month
- Species
- Confidence level
- Time of day: Dawn, Daylight, Dusk, Night

Additional criteria for search filtering should include:

- Rarity (requires user-configurable rarity listing)
- Capture length
- Number of detections per day
- User tags (requires implementing tagging feature)
- Weather conditions

What else?

jkrauska, 2024-12-19:
Filter out frequent birds. (I have some hummingbirds which take up MOST of the recordings)

billyvg, 2025-05-29:
I'm new to this project and just wondering if this ticket is still relevant? It seems the top half of these available here:

[image]

Does the issue involve updating the existing UX? Or should the issue be updated to only require the remaining items that are not available?

```
Rarity (requires user-configurable rarity listing)
Capture length
Number of detections per day
User tags (requires implementing tagging feature)
Weather conditions
```

tphakala, 2026-02-14:
Closing as largely addressed. The Svelte-based frontend now includes filtering by species, confidence, time of day, and date range in the detection views. As a commenter noted, most of the requested features are now available. If specific browsing capabilities are still missing, please open a targeted issue.