tphakala/birdnet-go #323: FFMpeg Problem

pack2010, 2024-11-22:
So I got birdnetgo working great but then had to replace my camera that I was using for audio input. Now I'm getting this error message when I run the program.

2024/11/21 22:21:15 Starting ffmpeg with command:  C:\ffmpeg\bin\ffmpeg.exe -rtsp_transport tcp -i rtsp://xxxx@xxx.xxx.x.xxx/Preview_01_main -loglevel error -vn -f s16le -ar 48000 -ac 1 pipe:1
2024/11/21 22:21:15 Error reading from ffmpeg: EOF
2024/11/21 22:21:15 FFmpeg wait error: exit status 0xcecfcb08
2024/11/21 22:21:18 Restarting FFmpeg.

Trying to figure out what happened and I'm at a loss. Thanks in advance for any help!

tphakala, 2024-12-18:
Can you try with latest nightly build? https://github.com/tphakala/birdnet-go/releases/tag/nightly-20241218

pack2010, 2024-12-19:
Hey, it actually works now! Thanks so much!

jkrauska, 2025-01-01:
@tphakala this looks ok to close now.

closed 2025-01-01 by tphakala