tphakala/birdnet-go #324: error when start Birdnet-go container

jhackeur, 2024-11-25:
## Issue Type
- [ ] Bug report

## Description
[A clear and concise description of the issue or feature request]

## Steps to Reproduce (for bugs)
Start docker instance and see logs:
root@vl77-ubuntu-srv-lab:/home/administrateur# docker run ghcr.io/tphakala/birdnet-go
Created default config file at: /root/.config/birdnet-go/config.yaml
BirdNET-Go build date: 2024-06-09T12:51:19Z, using config file: /root/.config/birdnet-go/config.yaml
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 4 threads of available 4 CPUs
2024/11/25 09:09:54 Using latest range filter model
System details: linux debian 12.5 on unknown hardware
Starting analyzer in realtime mode. Threshold: 0.8, overlap: 1.5, sensitivity: 1, interval: 15
Logging disabled
2024/11/25 09:09:54 Clip retention policy: usage
⇨ http server started on [::]:8080
Available Capture Sources:
  0: Discard all samples (playback) or generate zero samples (capture), null
****2024/11/25 09:09:54 Error selecting capture source: no suitable capture source found for device setting sysdefault****

## Expected behavior
Container should start without issue, but exit with code 1

## Actual behavior
Exit with code 1:
root@vl77-ubuntu-srv-lab:/home/administrateur# docker ps -a
CONTAINER ID   IMAGE                         COMMAND                  CREATED              STATUS                          PORTS                                                           NAMES
55e3e2761b48   ghcr.io/tphakala/birdnet-go   "/usr/bin/birdnet-go…"   About a minute ago   Exited (1) About a minute ago                                                                   brave_babbage

## Environment
### Version Information (build date)
Ubuntu on ESX running last version of docker

### System Information
### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes, same issue

tphakala, 2024-11-26:
Please see installation instructions on how to run Docker version https://github.com/tphakala/birdnet-go/blob/main/doc/installation.md

jhackeur, 2024-11-26:
Yes this is the guide I follow...

If I run this command as mentionned in the guide:
docker run -ti -p 8080:8080 --device /dev/snd ghcr.io/tphakala/birdnet-go:latest

I got the issue

rammytheone, 2024-12-04:
What sound card and microphone do you have? Is it available in Ubuntu? In Proxmox I had to assign a sound card to a virtual machine.

jhackeur, 2024-12-04:
@rammytheone purpose is to use RSTP stream, so no microphone.
I have vmware sound card.

I have installed this birdnet-go program directly, without using docker, and it works well. So i will stay like this

simongis, 2025-02-07:
Is it possible to install BirdNetGO without an audio device?  I only have a RTSP for my feed and struggling to get up and running

tphakala, 2025-02-21:
It is now possible to start BirdNET-Go without any configured audio sources, please update to latest nightly or v0.6.2 release.

closed 2025-02-21 by tphakala