tphakala/birdnet-go #337: Daily Summary fails to display from MySQL database

HWittingen, 2024-12-20:
## Issue Type
- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
When MySQL database is configured, the Daily Summary does not populate.

## Steps to Reproduce (for bugs)
1. Configure for MySQL
2. Have a fully populated database
3. Launch the BirdNET-Go main webpage 

## Expected behavior
A 15 species detailed Daily Summary

## Actual behavior
Empty Daily Summary

## Screenshots
2024/12/20 19:25:36 /home/pi/src/BirdNET-Go/internal/datastore/interfaces.go:200 Error 1055 (42000): Expression #2 of SELECT list is not in GROUP BY clause and contains nonaggregated column 'birdnet-go.notes.scientific_name' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by
[51.441ms] [rows:-] SELECT common_name,scientific_name,COUNT(*) as count FROM `notes` WHERE date = '2024-12-20' AND confidence >= 0 GROUP BY `common_name` ORDER BY count DESC LIMIT 30
ERROR: 2024/12/20 19:25:36 handlers.go:65: Error: Failed to get top birds data: Error 1055 (42000): Expression #2 of SELECT list is not in GROUP BY clause and contains nonaggregated column 'birdnet-go.notes.scientific_name' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by (Code: 500, Underlying error: Error 1055 (42000): Expression #2 of SELECT list is not in GROUP BY clause and contains nonaggregated column 'birdnet-go.notes.scientific_name' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by)

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [X] PC/Server Windows 11, BirdNET-Go (Windows ARM64 build today)
  - [X] Linux Fedora fc40, MySQL server 8.0.40
,  - OS: [e.g. Raspberry Pi OS, Ubuntu, Debian]
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [ ] No
- [X] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

tphakala, 2024-12-20:
Can you test if this commit fixes this? https://github.com/tphakala/birdnet-go/commit/5323e450ef056a5d1b81c822bfadcaa91e98c275

HWittingen, 2024-12-20:
Issue solved. Thanks.

--

met vriendelijke groet,

Henk Wittingen - Ede gld - NL

(verzonden met Outlook Microsoft 365 en Windows 11)

All incoming and outgoing mail is scanned for viruses and spam
using Amavis, ClamAV and Spamassassin invoked from Postfix.

Can you test if this commit fixes this? 5323e45<https://github.com/tphakala/birdnet-go/commit/5323e450ef056a5d1b81c822bfadcaa91e98c275>

tphakala, 2024-12-20:
fixed by https://github.com/tphakala/birdnet-go/commit/5323e450ef056a5d1b81c822bfadcaa91e98c275

closed 2024-12-20 by tphakala