tphakala/birdnet-go #350: Birdnet-Go Not Accessible after Upgrade to '25' or '25-2'

brooksben11, 2024-12-26:
## Issue Type
- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
After upgrading both to '25' and '25-2', the webpage for Birdnet-Go is inaccessible. Loading a backup to a previous version fixes all issues. This is what I see in my logs when I try to access the dashboard:
`2024/12/26 12:33:55 [error] 3423#3423: *1 connect() failed (111: Connection refused) while connecting to upstream, client: 172.30.32.2, server: db21ed7f-birdnet-go, request: "GET / HTTP/1.1", upstream: "http://127.0.0.1:8080/", host: "192.168.1.10:8123", referrer: "http://192.168.1.10:8123/hassio"
2024/12/26 12:33:55 [error] 3423#3423: *1 connect() failed (111: Connection refused) while connecting to upstream, client: 172.30.32.2, server: db21ed7f-birdnet-go, request: "GET / HTTP/1.1", upstream: "http://127.0.0.1:8080/", host: "192.168.1.10:8123", referrer: "http://192.168.1.10:8123/hassio"
[26/Dec/2024:12:33:55 -0600] 502 192.168.1.86, 172.30.32.1(172.30.32.2) GET / HTTP/1.1 (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36)`

## Steps to Reproduce (for bugs)
1. Upgrade to '25' or higher
2. Dashboard inaccessible

## Expected behavior
For it to work. :)

## Actual behavior
It doesn't work.

## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

### System Information
x86-64 laptop running HAOS.

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

tphakala, 2024-12-26:
I don't know what those version numbers are, also those log messages are not from BirdNET-Go

brooksben11, 2024-12-26:
My mistake, I'm using a downstream fork and accidently ended up on yours. :)

closed 2024-12-26 by brooksben11