tphakala/birdnet-go #357: install.sh expects S16_LE capable recording

jkrauska, 2024-12-30:
## Issue Type
- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
[A clear and concise description of the issue or feature request]
install.sh script expects audio device to support S16_LE.
Not all audio inputs support this.  eg. BOYA BY-LM40
It only support Signed 24 bit Little Endian in 3bytes. S24_3LE mode.

## Steps to Reproduce (for bugs)
1. use install.sh on a pi with a BOYA BY-LM40

## Expected behavior
USB audio device works

## Actual behavior
install fails
## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
current dev release

```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

### System Information
- Device: 
  - [x ] Raspberry Pi 5

- OS: Raspberry Pi OS
- OS Version: Raspberry Pi OS Bullseye
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 5]

  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: BOYA LM-40
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [x ] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
unclear if later code will require SE16_LE.. haven't see arecord calls in go.

## Possible Solution
try other formats?

jkrauska, 2024-12-30:
on deeper inspection, we assume its S16 in two places

https://github.com/search?q=repo%3Atphakala%2Fbirdnet-go%20malgo.FormatS16&type=code

lets see if the malgo package has the ability to DETECT the encoding.

tphakala, 2024-12-30:
BirdNET model expects input audio to be in 16-bit 48khz mono format

jkrauska, 2024-12-31:
We should include that somewhere I guess.  If I bypass the check, the app works, but it thinks all the birds sounds are humans.  I'm going to benchmark how much load dynamically re-encoding takes and try to add some additional error handling if the hardware doesn't support 16-bit.