tphakala/birdnet-go #360: docker buildx missing

jkrauska, 2024-12-30:
## Issue Type
- [x] Bug report
- [ ] Feature request
- [x] Question / Support

## Description
I'm trying to use a rpi5 as a dev platform and the docker installed by install.sh does not include buildx.

Thus when I try to start a dev container on the rpi5 via remote, I get this error in the docker setup.
[2024-12-30T06:25:34.214Z] docker: 'buildx' is not a docker command.

## Steps to Reproduce (for bugs)
1. try to setup dev environment on rpi5
2. [Second Step]
3. [and so on...]

## Expected behavior
[What you expected to happen]
runtime docker used on rpi5 would include buildx or more documentation in the developer notes on getting buildx.

## Actual behavior
unable to start dev container remotely using vscode on a rpi5.

## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)
- OS: [e.g. Raspberry Pi OS, Ubuntu, Debian]
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

jkrauska, 2024-12-30:
seems I need docker engine, not just docker.io.
https://docs.docker.com/engine/install/debian/

will dig in to this later

tphakala, 2024-12-30:
There are couple solutions, I personally have installed latest version of docker on my dev system from Docker repository, this includes buildx. Instructions are at https://docs.docker.com/engine/install/debian/

If you want to stick with Debian/rpi os repository you need to install buildx plugin

```
apt install docker-buildx-plugin
docker buildx install
```

jkrauska, 2024-12-31:
docker-buildx-plugin is not part of the stock debian/rpi os.

```bash
jkrauska@birdnet:~ $ sudo apt install docker-buildx-plugin
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package docker-buildx-plugin
jkrauska@birdnet:~ $ apt-cache search buildx
jkrauska@birdnet:~ $
```

Would you be open to be changing the install.sh docker steps to use the full docker engine?
This would ensure that the dev environment and the build environment are more similar and reduce friction for new developers.

With full docker engine install I was able to get much farther.   
Thank you.

closed 2024-12-31 by jkrauska