tphakala/birdnet-go #371: Daily summary list overflows UI element on Firefox

joriws, 2025-01-06:
## Issue Type
- [X] Bug report

## Description
UI space vs daily summary detection list, list overflows UI element at least on latest Firefox 133.0.3 (64-bit) in Windows 11. (green area). This is better with MS Edge browser (not overflowing UI element area) but not fully clean with rounded corner vs last detection line, should there be some rows of UI-element (white line). Recent detections has rounded corners, Daily Summary does not.

Secondary issue, can recording screenshot be smaller, it has a lot of black on top to compress user interface (red area). Or is this again Firefox issue but also Edge shows a lot of black?

## Steps to Reproduce (for bugs)
1. Open dashboard with Firefox

## Expected behavior
[What you expected to happen]

## Actual behavior
[What actually happened]

## Screenshots
[If applicable, add screenshots to help explain your problem]

Firefox screenshot
[image]

Edge screenshot, frequency image black areas on Edge.
[image]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```
🐦 BirdNET-Go build date: 2025-01-06T08:16:09Z, using config file: /root/.config/birdnet-go/config.yaml
BirdNET GLOBAL 6K V2.4 FP32 model initialized, using 1 threads of available 1 CPUs
System details: linux debian 12.8 on unknown hardware

Docker container release:
REPOSITORY                        TAG                                        IMAGE ID       CREATED         SIZE
ghcr.io/tphakala/birdnet-go       **cb9720145da771b194a0755b8dbc697959516947**   5726d7bab294   2 hours ago     653MB

### System Information
- Device: 
  - [X] Raspberry Pi (specify model, e.g., Raspberry Pi 3)

tphakala, 2025-01-13:
Regarding spectrogram, it shows frequency range from 0 Hz to 12 kHz, upper frequency range is not black for everyone, here is example on how it looks with good microphone which has wide frequency range and high sensitivity.

[image]

In this example microphone is AOM5024 capsule based mic connected to Focusrite XLR interface.

joriws, 2025-01-13:
Ok I did not think of that. The RTSP-G.711-audiostream from security cam I am using is sampled 8kHz (max 3.4kHz audio freq). 

But spectogram renderer knows input frequency so it could limit the graph based on sample rate? In my mind the black top is not the issue, I was just thinking on compressing UI for small devices like phones. Like a lot of white space between detections.

Is there any effect on AI-model with limited frequencies coming from mic? Because RTSP webcam outside is just so convenient audio source.

tphakala, 2025-01-13:
Overflow issue fixed with https://github.com/tphakala/birdnet-go/pull/383

closed 2025-01-24 by tphakala