tphakala/birdnet-go #372: OpenWeather language setting now ignored

HWittingen, 2025-01-06:
## Issue Type
- [x] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
The today refactor of the Weather Integration break the configured OpenWeather language setting.

## Steps to Reproduce (for bugs)
1. [First Step]
2. [Second Step]
3. [and so on...]

## Expected behavior
I'd expected the Dutch descriptions as configured in settings and stored in the database.

## Actual behavior
All weather desciptions are in English.

## Screenshots
[If applicable, add screenshots to help explain your problem]

## Environment
### Version Information (build date)
Provide the following line from your system:
```
$ ./birdnet-go
[Paste the output line here, which should look like this:]
BirdNET-Go build date: YYYY-MM-DDTHH:MM:SSZ, using config file: /path/to/config.yaml
```

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ ] PC/Server (please specify)
- OS: [e.g. Raspberry Pi OS, Ubuntu, Debian]
- OS Version: [e.g. Raspberry Pi OS Bullseye, Ubuntu 20.04.2 LTS]
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: [e.g. 4GB]
  - Storage type and size: [e.g. 32GB microSD card, 128GB SSD]
- Additional hardware (if applicable):
  - USB Audio device: [e.g. Brand, Model]
  - Other peripherals: [e.g. Specific microphone, GPS module]

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
Cause is the (hard)coded weather desription being deducted from the weather icon.

## Possible Solution
Take the desciption as stored in the database Iinstead of deriving it hardcoded from the icon.

HWittingen, 2025-01-06:
Language setting is not ignored only when displaying earlier stored weather descriptions.
The refactored code also ignores during the databse storage of the latest weather info.

tphakala, 2026-06-02:
This is fixed with frontend i18n