tphakala/birdnet-go #396: v0.6.0 has no linux/arm64/v8 image

d0ugal, 2025-01-16:
## Issue Type
- [x] Bug report
- [ ] Feature request
- [x] Question / Support

## Description

Firstly, congrats of the release. It looks great, thank you!

I tried to upgrade from 0.5.5 to 0.6.0 but I get this error from docker when I try to pull the image. It seems like 0.5.5 was built for arm64 but not 0.6.0.

`no matching manifest for linux/arm64/v8 in the manifest list entries`

Looking at [the package](https://github.com/tphakala/birdnet-go/pkgs/container/birdnet-go/337684097?tag=0.6.0) I see `unknown/unknown` which seems odd. 
[image]

### System Information
- Device: 
  - [x] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
- OS: Raspberry Pi OS
- OS Version:  Raspberry Pi OS Bookworm
- Hardware specs:
  - CPU: [e.g. Raspberry Pi 4 Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz]
  - RAM: 4GB
  - Storage type and size: 32GB microSD card
- Additional hardware (if applicable):
  - USB Audio device: Generic mic

isZumpo, 2025-01-17:
@tphakala a easy fix would be to add back my tagging logic:
https://github.com/tphakala/birdnet-go/commit/a5e3d4e4651a0d7b4150ce76b39a26eed63479af
which you removed here: https://github.com/tphakala/birdnet-go/commit/0a042c1d319e2c0b4797c7eb81bbd5f61531f831

There is no need to rebuild the container each night/release since it has already been built when the commit was merged. We can just add tags to the existing container.

d0ugal, 2025-01-26:
Thank you for the fix on the nightly tag. I've switched to using that for now. Would it be possible to release 6.0.1?

tphakala, 2025-02-21:
Problem is now resolved, there is v0.6.2 release with working images for both architechtures.

closed 2025-02-21 by tphakala