tphakala/birdnet-go #415: RTSP Stream restart not consistently happening on latest dev build

manicholls, 2025-01-25:
## Issue Type
- [X] Bug report
- [ ] Feature request
- [ ] Question / Support

## Description
[A clear and concise description of the issue or feature request]
Latest Dev build does not restart RTSP sources sometimes

## Steps to Reproduce (for bugs)
1. RTSP feed gets interrupted but the restart more often than not does not happen.
```
   2025/01/25 12:08:47 No data received from RTSP source rtsp://192.168.30.69:8080/stream.sdp for 60 seconds, triggering restart
```
## Expected behavior
Restart doesn't happen

## Actual behavior
The restart doesn't happen and data is no longer processed.

## Environment
### Version Information (build date)
Provide the following line from your system:
```./birdnet-go
🐦 BirdNET-Go v (built: 2025-01-15T20:45:29Z), using config file: /root/.config/birdnet-go/config.yaml
BirdNET-Go CLI
```

### System Information
- Device: 
  - [ ] Raspberry Pi (specify model, e.g., Raspberry Pi 4 Model B)
  - [ ] Other SBC (please specify)
  - [ X] PC/Server (please specify)
- OS: Debian with Docker
- OS Version: Debian 12
- Hardware specs:
  - CPU: Intel N100
  - RAM: 4GB
  - Storage type and size: 500GB SSD
- Additional hardware (if applicable):
  - USB Audio device: Pizerow streaming a usb mic via VLC (UDP RTSP)
  

### For Non-Raspberry Pi Linux Users
Have you tested with the latest dev branch container image?
- [ ] Yes
- [ ] No
- [ ] Not applicable (Windows/macOS user)

If not, first test with the latest dev branch container image and see if the issue is already fixed.
```
$ docker pull ghcr.io/tphakala/birdnet-go:dev
```

## Additional context
[Add any other context about the problem here, such as any custom configurations or specific usage scenarios]

## Possible Solution
[If you have a suggestion for fixing the bug or implementing the feature]

xconverge, 2025-02-06:
I am seeing this as well in 0.6.0, I will try the latest dev as well but not sure what to expect

xconverge, 2025-02-11:
I cant try the latest nightly because there seems to be a regression related to my setup/config of an RTSP only source

```
2025/02/11 14:56:04 /home/dev-user/src/BirdNET-Go/internal/datastore/interfaces.go:398 SLOW SQL >= 200ms
[3363.484ms] [rows:113] SELECT scientific_name FROM `notes` GROUP BY `scientific_name`
Logging disabled
HTTP server started on port 8080 (AutoTLS: false)
2025/02/11 14:56:04 Clip retention policy: age
⇨ http server started on [::]:8080
2025/02/11 14:56:04 ⬆️  Starting ffmpeg with command: /usr/bin/ffmpeg -rtsp_transport tcp -i rtsp://xxx.xxx.xxx.xxx:8554/mic -loglevel error -vn -f s16le -ar 48000 -ac 1 -hide_banner pipe:1
2025/02/11 14:56:04 Starting audio processing for RTSP source: rtsp://192.168.1.110:8554/mic (display name: Camera mic)
❌ Error selecting capture source: no audio devices available in container
Please map host audio devices by running docker with: --device /dev/snd
Instructions for running BirdNET-Go in Docker are at https://github.com/tphakala/birdnet-go/blob/main/doc/installation.md
Available Capture Sources:
  0: Discard all samples (playback) or generate zero samples (capture), null
2025/02/11 14:56:21 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:21 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:21 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:21 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
2025/02/11 14:56:24 [security/cloudflare] Cloudflare Access is disabled
```

tphakala, 2025-02-21:
The Dev image build workflow hasn't been working for a while, so it didn't include all the recent fixes. The image is now up to date, and RTSP stream restarts should now work properly.

closed 2025-02-21 by tphakala